FreeCAD/FreeCAD-Bundle #130: UI messaged up in latest weekly AppImage

ifohancroft, 2022-08-02:
As can be seen here https://youtu.be/M3rtcglKP9U the UI of the latest weekly AppImage is messed up. I have removed my configs and stuff to make sure they aren't causing the problem.

```
OS: Manjaro Linux (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.21.29865 (Git) AppImage
Build type: Release
Branch: master
Hash: fb5c731847398c5bfbc37468ba851e70860a86d5
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
```

Forum post: https://forum.freecadweb.org/viewtopic.php?p=614419

ifohancroft, 2022-08-07:
The bug is still present in today's AppImage as well:

```
OS: Manjaro Linux (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.21.29997 (Git) AppImage
Build type: Release
Branch: master
Hash: b52967d52ac46eff7c59e74d991f3f5b298944ef
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * MechanicalKeyboards
  * sheetmetal 0.2.49
  * kicadStepUpMod 10.11.9
```

ifohancroft, 2022-08-11:
I am closing the bug report as the problem turned out to be on my end. Apparently having installed KDE and/or Kvantum at some point has messed up things. The current value of `$QT_FONT_DPI` was 92, so executing `QT_FONT_DPI=96 freecad` fixed things. I need to check what I still have installed that sets the variable and remove it.

closed 2022-08-11 by ifohancroft

luzpaz, 2022-08-11:
Thanks for following up @ifohancroft  

@PLChris where can we mention this workaround, in 'known issues'?

ifohancroft, 2022-08-11:
@luzpaz Most welcome!
If I manage to find what is setting the environment variables, a workaround may not be needed. Instead, a heads up to look for that env (or the application setting it) in-case of UI problems might be worth it.

ifohancroft, 2022-08-11:
Here's what I found. It's a deep rabbit hole, so I'll try to only include what's relevant.

1. The `QT_FONT_DPI` and `QT_SCALE_FACTOR` environment variables. Long story short, in gsettings, MATE's org.mate.interface has a key called window-scaling-factor-qt-sync. If set to true (it was) it also sets the DPI for QT. It sets it correctly to 92 because my monitors are actually 92 DPI.
2. When setting the value to false, those two environment variables are no longer set, however FreeCAD still has messed up UI (and we know it works when we set the font dpi to 96 for QT), so QT probably agrees that the monitors' DPI is 92 (again, that's correct).
3. As we've already established, in-order for FreeCAD's AppImage mentioned above to work, the DPI for the QT font must be set to 96.

We have a working workaround now, however, keeping in-mind that just a couple of weeks ago the AppImages worked without this (custom compiling FreeCAD hasn't worked with me for awhile (same problem)) I am thinking either something in Qt changed or something in FreeCAD so I'm hoping we can find the root cause.

The latest AppImage that I have that works is this one:

```
OS: Manjaro Linux (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: English/United States (en_US)
Installed mods: 
  * MechanicalKeyboards
  * sheetmetal 0.2.49
  * kicadStepUpMod 10.11.9
```

The earliest I currently have, that doesn't work, is this one:

```
OS: Manjaro Linux (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.21.29997 (Git) AppImage
Build type: Release
Branch: master
Hash: b52967d52ac46eff7c59e74d991f3f5b298944ef
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * MechanicalKeyboards
  * sheetmetal 0.2.49
  * kicadStepUpMod 10.11.9
```

Does anyone have any AppImages between those two? I'd like to test which one is the last one that works and which one is the earliest that doesn't to see if the Qt version changed and hopefully narrow down the problem.

I'll hold off marking the forum post as fixed for now as I'm hoping we can find the root cause and I can write it there when marking it as fixed.

P.S. It used to be that when you go to download the weekly AppImage you have previous versions there as well, now it's just the latest one. What happened to that? Can we get back to having previous versions as well?

Btw when my AppImage stopped working, as I didn't expect that to happen, I had already deleted the previous one after downloading the new one (before testing it) and was almost left without a working FreeCAD :D