FreeCAD/FreeCAD-Bundle #131: osx-arm

looooo, 2022-08-03:
- [ ] FreeCAD logo not working
- [ ] dmg version name is wrong (the actual version is higher than 29436) bundling process needs to run on m1 directly

adrianinsaval, 2022-08-03:
Running in an automated way in an M1 might be hard to achieve right now (github doesn't have arm runners AFAIK), is this needed only to get info version?

looooo, 2022-08-03:
No it's also needed to compile freecad. There are some more issues with cross-compiling. There are some instructions to setup GitHub action runners locally. Maybe this would be an idea to further improve automatisation of osx-arm64 and linx aarch64.

looooo, 2022-08-28:
at the moment I do the osx-arm builds and bundles locally, waiting for GitHub-action to offer osx-arm64 support.

closed 2022-08-28 by looooo

sammcj, 2023-03-22:
Any updates to this? It's still an issue and the only decent performing build is that single arm build from December.

JohnOCFII, 2023-03-22:
> Any updates to this? It's still an issue and the only decent performing build is that single arm build from December.

Watch this thread.  There is now a working Conda build as of today.  https://forum.freecad.org/viewtopic.php?t=76777&start=20

sammcj, 2023-03-22:
That's awesome news! Thank you! 😄