FreeCAD/FreeCAD-Bundle #134: Appimage not signed

leoheck, 2022-09-10:
I am upgrading the Freecad Weekly with the Appimage upgrade tool.
This is happening.. last week it was ok.
[image]

luzpaz, 2022-09-10:
Related #6

adrianinsaval, 2022-10-04:
> Last week I did not have this issue.

that's strange, AFAIK the appimage never got signed.

leoheck, 2022-10-04:
Exactly what I thought. But at that time, the installation did not complained and the progress bar turned green instead of yellow. Maybe it was a different version of the updater tool then. Unfortunately I cannot check this anymore. I can only get the current version I am using.

adrianinsaval, 2024-08-27:
duplicate of #6 and this should be solved for x86_64 now

closed 2024-08-27 by adrianinsaval