FreeCAD/FreeCAD-Bundle #137: SSL: pip issue in FreeCAD_weekly-builds conda-Windows-py310

easyw, 2022-10-11:
following this issue @forum 
https://forum.freecadweb.org/viewtopic.php?p=632089#p632089

> I'm trying to use pip to install assembly 3 solver in
> github FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/
> FreeCAD_weekly-builds-30492-2022-10-09-conda-Windows-x86_64-py310
> (I have no issue with py3.8 version)
> 
> It seems pip doesn't get SSL module: 

```
C:\FreeCAD\bin>scripts\pip install  py_slvs
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/py-slvs/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/py-slvs/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/py-slvs/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/py-slvs/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/py-slvs/
Could not fetch URL https://pypi.org/simple/py-slvs/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/py-slvs/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
ERROR: Could not find a version that satisfies the requirement py_slvs (from versions: none)
ERROR: No matching distribution found for py_slvs
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping

```

adrianinsaval, 2022-10-12:
probably also related: https://forum.freecadweb.org/viewtopic.php?f=3&t=72381&hilit=ssl

easyw, 2022-10-19:
yes

closed 2022-10-21 by easyw

luzpaz, 2022-10-21:
30694 ?  
Could you please provide a link? TIA

adrianinsaval, 2022-10-21:
https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-30694-2022-10-20-conda-Windows-x86_64-py310.7z

adrianinsaval, 2022-10-21:
possbily fixed by https://github.com/FreeCAD/FreeCAD-Bundle/commit/e53d62b0eaa4bea8045782b379d24600e7206e0c

Hueristic, 2024-05-26:
Release build FreeCAD_weekly-builds-37464

Py3.10 Wx64