FreeCAD/FreeCAD-Bundle #138: Current build is crashing when creating Bodies

leoheck, 2022-10-22:
How can download the latest working build? I don't have the one from the last week anymore =/

Related conversation is here: https://forum.freecadweb.org/viewtopic.php?f=3&t=72811

leoheck, 2022-10-22:
Ah, I found out that the the nightly snap version is working.

luzpaz, 2022-10-22:
Related to https://github.com/FreeCAD/FreeCAD/issues/7628 ?

leoheck, 2022-10-22:
This happens when I do the following:

- New Body
- New Sketch
- Draw something (I was drawing 2 concentric circles)
- Close the Sketch (it may break now, I could reproduce this most of the time)
- If it works, then make a pad (it may break here too, 1 or 2 times it has broken here)

luzpaz, 2022-10-26:
About info ?

leoheck, 2022-10-26:
[image]

luzpaz, 2022-11-04:
Can you restest with the latest master ?

leoheck, 2022-11-04:
Looking good
```
[code]
OS: Ubuntu 22.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.30811 (Git) AppImage
Build type: Release
Branch: master
Hash: c490b93173ff90f7ed8de270d674530d5995446e
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.0
  * kicadStepUpMod.backup1667488881.9588385 10.15.1 (Disabled)
  * A2plus 0.4.60
  * Curves 0.5.13
  * fasteners 0.4.22
  * lattice2 1.0.0
  * Assembly4 0.12.4
  * kicadStepUpMod 10.16.5
  * Help 1.0.3
  * Assembly4.backup1667488875.2401915 0.12.4 (Disabled)
[/code]
```

closed 2022-11-04 by leoheck