FreeCAD/FreeCAD-Bundle #139: FreeCAD_0.21.29457-2022-10-22-conda-Linux-x86_64-py310.AppImage in Ubuntu 18.04: failed to  get asm3 to work

CBtiger, 2022-10-24:
Trying to run the new AppImage on Ubunut 18.04 I could not get Asm3 working. It did work in FreeCAD_0.19-24276-Linux-Conda_glibc2.12-x86_64.AppImage. Also I got strange output on the Python console. 
When I repeated everything with my .local/share/FreeCAD folder removed (fresh install) I did not get the DEBUG outputs on the Concole, but Asm3 failed with the exact the same symptoms.

**1 Python Console output at start**
```
10:18:34  During initialization the error "No module named 'Drawing'" occurred in /home/hans/.local/share/FreeCAD/Mod/Animation/InitGui.py
10:18:34  Please look into the log file for further information

**2 Python Console output when using the Addon Manager** 
```
10:19:37  Addon Developer Warning: Repository branch set in package.xml file for addon FreeCAD documentation (main) does not match the branch it was fetched from (master)
10:19:37  Addon Developer Warning: Repository branch set in package.xml file for addon Trails (main) does not match the branch it was fetched from (master)
10:19:37  DEBUG:git.cmd:Popen(['git', 'pull', '--ff-only'], cwd=/home/$USER/.cache/FreeCAD/Cache/AddonManager/Macros, universal_newlines=False, shell=None, istream=None)
10:19:38  DEBUG:git.util:Failed checking if running in CYGWIN due to: FileNotFoundError(2, 'No such file or directory')
10:19:38  DEBUG:git.cmd:Popen(['git', 'fetch'], cwd=/home/$USER/.local/share/FreeCAD/Mod/Assembly3, universal_newlines=False, shell=None, istream=None)
10:19:38  DEBUG:git.cmd:Popen(['git', 'status'], cwd=/home/$USER/.local/share/FreeCAD/Mod/Assembly3, universal_newlines=False, shell=None, istream=None)
10:20:07  DEBUG:git.cmd:Popen(['git', 'fetch'], cwd=/home/$USER/.local/s
```

**3. Assembly 3 udapte/installation**
I was trying to update Assembly3. I did **not** get the selection window but it installed py-slvs with pip as I saw in the console (see below). But it poped up a small failed dialog with OK button. 
```
10:22:22  Looking in indexes: http://localhost:3141/root/pypi/+simple/
Collecting py-slvs
  Downloading http://localhost:3141/root/pypi/%2Bf/f51/3627031c5da9d/py_slvs-1.0.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (294 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 294.1/294.1 kB 374.5 MB/s eta 0:00:00
Installing collected packages: py-slvs
Successfully installed py-slvs-1.0.3
```
Indeed, asm3 did not work. The "Assembly" icon in the tree was missing and when pressing the Solve button on a simple test scene the output is this
```
10:19:18  48.891045 <asm3.sys> solver.py(412): no assembly found
10:19:22  4.087824 <asm3.sys> solver.py(412): no assembly found
```

I uninstalled it then installed it again. Now I got the Python Module choice dialog and selected install for py-slvs. It asked for restart. After the restart, asm3 still did not work and showed the symtopms. 

On a fresh install asm still did not work with the same symptoms
* AppManage did not install py-lsvs on first install. Asked for restart. Asm3 did not work
* After Uninstall/Install the instal-python dialog came up. It asked for restart. Ams3 did not work

Version Info
[code]
OS: Ubuntu 18.04.6 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.29457 (Git) AppImage
Build type: Release
Branch: development/toponaming
Hash: 18ff9bcf2610a04e07b87fa077abedeb4b2b2761
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * TabBar
  * dodo
  * Assembly3 0.11.3
  * DynamicData 2.46.0
  * Assembly4 0.12.3
  * Defeaturing
  * kicadStepUpMod 10.14.5
  * ExplodedAssembly
  * Manipulator 1.4.9
  * MeshRemodel 1.8919.0
  * ArchTextures
  * Trails 2022.1.0
  * ThreadProfile 1.84.0
  * Help 1.0.3
  * SelectorToolbar
  * Behave-Dark-Colors 0.1.1
  * LCInterlocking
  * InventorLoader 1.3.0
  * InventorUbuntuInstallScript
  * 3D_Printing_Tools
  * PieMenu 1.1.10
  * Animation
  * workfeature
  * lattice2 1.0.0
  * MakerWorkbench
  * Reporting
  * Curves 0.5.2
  * sheetmetal 0.2.58
  * IconThemes
  * fasteners 0.4.1
  * Render 2022.2.0
  * PreferencesPresets
  * Design456 0.0.1
  * exportOffline.sh
  * Pyramids-and-Polyhedrons
  * CurvedShapes 1.0.4
  * offline-documentation 1.0.0-alpha
  * Glass
  * Sample
  * fcgear 1.0.0
  * FCGear
  * BIM 2021.12.0
[/code]

CBtiger, 2022-10-24:
also posted and issue with Asm3
 [https://github.com/realthunder/FreeCAD_assembly3/issues/1282](url)

CBtiger, 2022-10-26:
The cause of the mesassage was that in Assembly Properties the Solver was set to "None". That also caused the missing tree icon. 
Setting the Solver to "SolverSpace" the Icon appeared and the WB worked OK.

The solver is saved with the file. This would cause the message to appear on a OK installation. Or vice versa, if you open a OK file on a Installation that is missing the python binding and the file is saved, it will produce this error - even when opened on an OK installation. 

The issue showed also that AddonManager saves the python binding files to "AdditionalPythonPackages" in the users "Mod" folder. This is not good, because different python version may be used. It would be better for AddOnManager to not install this binding, because the Asm3 WB itself will install it if it finds it missing. It uses a different path (in Linux Ubuntu/debian) .include/lib/site-packages/pythonXX/py_slvs. That way different python version will exist next to each other. This can be useful if different AppImages shall be used, e.g. for testing.

For more info see the linked Asm3 issue.

luzpaz, 2022-10-26:
If this is an Addon Manager issue...I suggest we open a ticket on https://github.com/FreeCAD/FreeCAD-addons/

CBtiger, 2022-10-26:
I am happy to do that, but I want to wait what realthunder will say.

CBtiger, 2022-10-30:
> If this is an Addon Manager issue...I suggest we open a ticket on https://github.com/FreeCAD/FreeCAD-addons/

https://github.com/FreeCAD/FreeCAD-addons/issues/243
done

closed 2022-10-30 by CBtiger