FreeCAD/FreeCAD-Bundle #140: How do you differentiate versions?

leoheck, 2022-10-28:
The first icon from the left to the right is the Snap version.
The others are AppImages.
They don't have an icon, they don't have a meaningful name nothing to differentiate, at least in the launcher as I am showing. 
Every time I use the AppImage update tool, it adds a new unknown icon there in this grid. 
How do I know which one is the latest one?
Is it possible to remove the old versions when using the AppImage tool, automatically?
Or at least, is it possible to add a version in the name that appears in this launcher so the user can detect which one is the latest one?

[image]

It would improve a bit if it could have the rev win the name, so instead of "Freecad Weekl..." it could show 
- Freecad 752
- Freecad 752 W...

luzpaz, 2022-11-06:
Good idea.

adrianinsaval, 2022-12-07:
The appimage could be replaced automatically by appimage update if it had always the same name, but that may not be desired either as we want the version information on the filename, I use this script to update my appimage:
```
#!/bin/sh

cd $HOME/Applications
$HOME/Applications/appimageupdatetool-x86_64.AppImage FreeCAD.AppImage && \
wasUpdated=$(find -name 'FreeCAD*weekly*.AppImage' | wc -l) && \
if [ $wasUpdated -eq 1 ]; then
   echo "Renaming to FreeCAD.AppImage"
   find -name 'FreeCAD*weekly*.AppImage' -exec mv {} FreeCAD.AppImage \;
fi
```

adrianinsaval, 2023-04-12:
@leoheck I recently discovered that appimage update tool has an --overwrite option which jjust replaces teh existing appimage instead of creating a new one when updating, I suggest you rename your appimage to FreeCAD-weekly.appimage, remove the old versions causing you trouble and tehn just use this option when updating.

Does that satisfy your needs? if yes I think this issue can be closed.

leoheck, 2023-04-13:
I have a script that update appimages. It is similar to yours but for ALL my appimages. Now which the icon fixed, it is easier to identify which Freecad is the one from here since it has a modified icon. 

Why don't you use AppImage loader it works pretty well.

closed 2023-04-13 by leoheck