FreeCAD/FreeCAD-Bundle #141: bug - 0.20.1 (macos 13.0) - display -> colors -> simplecolor does not work

TomMuc1, 2022-11-05:
bug - 0.20.1 (macos 13.0) - display -> colors -> simplecolor does not work

TomMuc1, 2022-11-08:
dmg image. screenshot for whatever that is supposed to be useful
[image]

luzpaz, 2022-11-08:
@TomMuc1 please post About info as requested above.

@ipatch sorry to impose, can you test this on homebrew and if possible on the appimage to see if you can reproduce ?

TomMuc1, 2022-11-08:
sorry!

[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: C/Default (C)
[/code]

luzpaz, 2022-11-08:
@PLChris do you mind also testing if you can reproduce on macos ?

ipatch, 2022-11-08:
> @ipatch sorry to impose, can you test this on homebrew and if possible on the appimage to see if you can reproduce ?

i'm not familiar with _simplecolor_ as i do not know what that is suppose todo. also the text in the posted screenshot is a language that i cannot read.

what is the intended desire/purpose of _simplecolor_? and what are the steps i use to reproduce on my end?

TomMuc1, 2022-11-08:
for those who never heard about deepl.com or google.com - a screenshot in english language setting
[image]

luzpaz, 2022-11-08:
>for those who never heard about deepl.com or google.com - a screenshot with english language setting 

@TomMuc1 no need to be flippant. We're helping you, right? You also created steps for us to jump through instead of providing a thorough bug report.

TomMuc1, 2022-11-10:
don't need help. just set both colors of the gradient to the same gray. btw installed the same freecad version on an old mac computer running mojave -  same error. so it might be a bug on many (all) macs.