FreeCAD/FreeCAD-Bundle #143: loading tracks error on LinkBranch and TNP V1

easyw, 2022-11-25:
I'm having an issue in loading tracks with FC TN v1 and Link Branch (fcad_pcb)
here the error I get:
```19:02:40  making tracks...
19:02:40    making 351 tracks  of width 0.60, (0/2880)
19:02:40  Traceback (most recent call last):
  File "C:/Cad/Progetti_K/3D-FreeCad-tools/fcad_pcb_minimal.py", line 21, in <module>
    pcb.makeTracks(shape_type='face',fit_arcs=True,thickness=0.05,holes=False)
  File "d:fcad_pcb\kicad.py", line 1840, in makeTracks
    objs.append(func(edges,label=label))
  File "d:fcad_pcb\kicad.py", line 1785, in _face
    return _wire(edges,label,True)
  File "d:fcad_pcb\kicad.py", line 1782, in _wire
    return _line(edges,label,width*0.5,fill)
  File "d:fcad_pcb\kicad.py", line 1777, in _line
    wires = findWires(edges)
  File "d:fcad_pcb\kicad.py", line 341, in findWires
    return [Part.Wire(e) for e in Part.sortEdges(edges)]
  File "d:fcad_pcb\kicad.py", line 341, in <listcomp>
    return [Part.Wire(e) for e in Part.sortEdges(edges)]
<class 'Part.OCCError'>: class StdFail_NotDone BRep_API: command not done
```
and the code I use:
```
import PySide
from PySide import QtGui, QtCore
import sys,os
sys.path.append(r"d:")
import fcad_pcb
from fcad_pcb import kicad
import importlib
importlib.reload(kicad)
from kicad_parser import KicadPCB
filename= r"D:\Temp\infinitas.kicad_pcb" #"Your-full-path-to-kicad-pcbnew-file"
pcb = kicad.KicadFcad(filename,via_skip_hole=False,via_bound=1)
pcb.setLayer(0) #'F.Cu')
# pcb.makeZones(shape_type='face',thickness=0.05, fit_arcs=True,holes=True)
#coppers = pcb.makeCoppers(shape_type='solid', holes=True, fuse=True)
pcb.makeTracks(shape_type='face',fit_arcs=True,thickness=0.05,holes=False)
# pcb.makeBoard()
```
and the test file:
[infinitas-pcb.zip](https://github.com/FreeCAD/FreeCAD-Bundle/files/10093752/infinitas-pcb.zip)

there is an issue closed here:
https://github.com/realthunder/fcad_pcb/issues/61
but it is closed but still with an open issue

realthunder, 2022-11-25:
Can you please try my latest snap release and see if it is fixed?

easyw, 2022-11-26:
@realthunder 
I'm on Mint, so I don't have snap...
is there an AppImgage uptodate available with the latest patch?

adrianinsaval, 2022-12-07:
I don't think this is the appropriate place for this issue

luzpaz, 2022-12-11:
Do we have an app image for Toponaming branch?

adrianinsaval, 2022-12-11:
yes but I think it's outdated now: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/topo-naming

realthunder, 2022-12-12:
I'll build a new one soon.

luzpaz, 2023-02-27:
Can we close this ticket ?

closed 2023-02-27 by realthunder