FreeCAD/FreeCAD-Bundle #144: macOS arm build app icon is blank

sslupsky, 2022-12-13:
A note to mention that the macOS arm build has a blank (all white) icon.
[image]

luzpaz, 2023-01-14:
@sslupsky please re-download macos conda and report back

luzpaz, 2023-01-14:
This should be fixed. Closing.

sslupsky, 2023-01-16:
Downloaded and looks good.  Thank you.

sslupsky, 2023-01-18:
@luzpaz Actually, there is still a minor problem.  The icon in the dock still appears blank.

[image]

luzpaz, 2023-01-19:
@sslupsky any idea why?

sslupsky, 2023-01-19:
@luzpaz Not sure.  I removed the icon from the dock and it appeared blank when I opened FreeCAD.  I rebooted, hoping to clear some cache somewhere, but that did not fix the issue.  I'll have to dig around.  It may be an artifact of the earlier problem where something is cached someplace but I don't really know where to look for it at the moment.

sslupsky, 2023-01-19:
@luzpaz Rebooting into Safe Mode seems to enough to clear the dock icon cache ... the icon is visible now.

closed 2023-01-19 by sslupsky

JohnOCFII, 2023-01-20:
FYI - the dock icon (as well as the icon in the Application folder) look fine for me with this version:
```[code]
OS: macOS 13.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.31483 (Git)
Build type: Release
Branch: master
Hash: 588620fd9fd138892b1cd2466f5c87d441493174
Python 3.11.0, Qt 5.15.6, Coin 4.0.0, Vtk 9.2.2, OCC 7.6.3
Locale: C/Default (C)
[/code]
```