FreeCAD/FreeCAD-Bundle #146: Support for 3DConnexion Devices in ARM Version?

steffenxx, 2022-12-16:
Apparently there is no 3DConnexion support implemented in the ARM version like in the x86 version. Is this planned?

luzpaz, 2022-12-20:
Please open discussion on the forum and x-post back here

steffenxx, 2022-12-20:
https://forum.freecadweb.org/viewtopic.php?f=3&t=74534

steffenxx, 2023-01-08:
I have already opened a discussion on 17 Dec. (See link) Unfortunately, so far without success.

adrianinsaval, 2023-01-17:
for clarity this issue is in relation to the macOS arm version, in the forum @looooo mentioned that we needed a download link for an ARM driver, looking here: https://3dconnexion.com/us/wp-content/uploads/sites/27/2023/01/Release_Notes-Mac-20230109.pdf it seems they have ARM support since January 2022, from version 10.7.0 if I'm reading it correctly, currently we are using the older 10.6.6 so it might work by just switching to the latest release: https://download.3dconnexion.com/drivers/mac/10-7-4_C5CD2A0D-58B6-4A8F-AFD1-54D5EB1BD29C/3DxWareMac_v10-7-4_r3493.dmg

adrianinsaval, 2023-01-18:
It's not about you updating the driver, we need the correct driver at compile time

> Maybe there are not enough Mac users with such devices.

More importantly, there aren't many developers using macOS to begin with then I don't know of someone using 3d mouse plus they'll have to ve using arm... It's a very niche case
@chennes do you happen to have a 3d mouse to try and get it working on your Mac?

steffenxx, 2023-01-18:
I understand that. Since Apple no longer builds x86 Macs, the ARM version will become more and more important in the future. Is there anything I can do to help ?

adrianinsaval, 2023-01-18:
Can you confirm that the link I posted contains a driver for the ARM architecture too?
Are you able to build FreeCAD in your machine in order to test it?

chennes, 2023-01-18:
I don't own one, but TBH I have been pondering their merits for some time. Maybe this can be my excuse 😄 . Which one should I get?

steffenxx, 2023-01-18:
I use the Space Mouse Wireless and the CAD Mouse Wireless. There was once a bundle to buy. I find the handling of the devices very good.

steffenxx, 2023-01-18:
> Can you confirm that the link I posted contains a driver for the ARM architecture too?
> Are you able to build FreeCAD in your machine in order to test it?

The release notes do not explicitly mention the ARM architecture. There is only a reference to the macOS version and the supported devices. But I can say that all versions since 10.7.0 basically run on MAC Studio and are supported by products like Prusa Slicer. 

I do not have a development environment installed. What would I need to create a FreeCad build?

adrianinsaval, 2023-01-18:
Hard for me to answer since I'm not a mac user, in the interest of testing as closely to our packaging environment it would be ideal for you to build on a conda environment, perhaps looo can provide better instructions.
My understanding is that you need to install conda, then create an environment for FreeCAD, install all it's dependencies and then you can build it inside that environment. Have look at https://wiki.freecad.org/Conda and here: https://github.com/looooo/freecad-feedstock/blob/d5cb2fd87de08e5b29b2ab4cd07448cefb6d898a/recipe/meta.yaml#L50-L95 for the dependencies and here: https://github.com/looooo/freecad-feedstock/blob/development/recipe/build.sh for a reference on building it

looooo, 2023-01-19:
if you want to build freecad locally and want to test things it's best to use `conda debug`. So you first get the freecad-feedstock repo, then you `cd` into the feedstock repo and call ` conda debug . -m .ci_support/osx_arm64_python3.11.____cpython.yaml`. At the end of this step there are some commands printed which you have to copy and paste to activate the build-env. Once the build env is activated you can call the script to build freecad (`bash conda_build.sh`)

if you want to use your own branch you need to change the directory in the `conda_build.sh` file.

steffenxx, 2023-01-21:
Thanks for the tips. I'm afraid this is a bit beyond what I can do, not being a developer myself. I was hoping that 3DConnexion support could be implemented in the ARM version in the same way as it is already in the x86 macOS. I currently have both versions on my Mac. With the current x86 version the SpaceMouse does not work with the latest ARM version. I wonder what I would have to do if I were to install the development environment on my machine? I would not be able to install the 3DConnexion support myself. My offer of help was rather aimed at that I would test versions if necessary. I would also donate a few EUR's via PayPal to a developer if he buys the devices and builds a full ARM version.

chennes, 2023-01-21:
I've ordered one, it should get here this week. I don't know if I'll get a chance to work on this before FreeCAD Day/FOSDEM, though, so there will be a bit of a delay.

looooo, 2023-01-22:
hey @chennes, I will be also at FOSDEM, maybe we can work on this if there is some time.

chennes, 2023-01-23:
@looooo do you have an ARM laptop? Mine is a desktop, so though I could bring the SpaceMouse, I can't easily bring the M1 :).

looooo, 2023-01-23:
Yes I have a M1 laptop which I bring with me.

steffenxx, 2023-02-18:
Is there already anything new ? :-)

chennes, 2023-02-18:
Well, I now have a SpaceMouse sitting next to my M2 laptop... I have the driver installed, but have not yet had a chance to see why it doesn't work in FreeCAD.

steffenxx, 2023-02-18:
It is already good that my problem can be reproduced. As I wrote above, my last working combination of driver and FreeCAD on macOS is FreeCAD for x86 and the old driver version V10.7.0.

chennes, 2023-02-18:
Are you self-compiled, or using one of the pre-built binaries?

steffenxx, 2023-02-18:
I use the binaries from this site.

chennes, 2023-02-18:
It's very strange, all of the setup code is working fine, we are making connection to the 3DConnexion daemon and connecting to its events, but the function that is supposed to be getting called just never is. I tried to take a look at how PrusaSlicer does it, but they use a totally different technique, apparently not using the 3DConnexion API at all (??).

chennes, 2023-02-19:
Well, the good news is that it can be made to work: I have a functional self-compiled spacemouse-enabled version on my M2 right not. The bad news is, I don't know why it's not working on the downloadable bundle. @looooo any clue here?

adrianinsaval, 2023-02-19:
as I understand it the current conda build script is completely skipping building this for arm right now:
https://github.com/looooo/freecad-feedstock/blob/b749bd16e65e7caac430aa88699464e693bfc206/recipe/build.sh#L26-L47

the script has to be updated to download the newer driver and compile this stuff for arm too

steffenxx, 2023-02-19:
> Well, the good news is that it can be made to work: I have a functional self-compiled spacemouse-enabled version on my M2 right not. The bad news is, I don't know why it's not working on the downloadable bundle. @looooo any clue here?

But it sounds like there will be a solution for an ARM version with 3DConnexion support soon. Great :-)

looooo, 2023-03-05:
does anyone know which version is osx-arm compatible?
https://3dconnexion.com/de/archived-drivers/

chennes, 2023-03-05:
I was able to use the latest version from their website, 10.7.4, on my M2.

steffenxx, 2023-03-11:
> Well, the good news is that it can be made to work: I have a functional self-compiled spacemouse-enabled version on my M2 right not. The bad news is, I don't know why it's not working on the downloadable bundle. @looooo any clue here?

Would it be possible for you to provide me with your working version? It would make me very happy. 

I like FreeCAD very much, but I also like my MAC. I am a bit disappointed that the last x86 version was updated 2 days ago and the ARM version was last updated on Jan 7.. Apple Silicon processors are the new standard for MAC's I hope FreeCAD will soon be maintained for the ARM processor by default.

adrianinsaval, 2023-03-11:
unfortunately the infrastructure is not place so that we can easily make regular arm releases as we do for other platforms, AFAIK the CI services we use still don't provide mac arm environments.

steffenxx, 2023-03-11:
> unfortunately the infrastructure is not in place so that we can easily make regular arm releases as we do for other platforms, AFAIK the CI services we use still don't provide mac arm environments.

Okay, I understand that. The number of ARM users will probably increase steadily and I really don't want to have to switch to another CAD software just because I have a modern computer. ;-)

chennes, 2023-03-11:
I have no idea if my builds are portable (I don't do any packaging). Can I just zip up my build directory?

luzpaz, 2023-03-12:
Aside: https://github.com/FreeCAD/FreeCAD-snap/issues/9 ARM builds for Snap packages (this is still a WIP)

adrianinsaval, 2023-03-12:
I suggest renaming this issue to make it clear it is about MacOS and not arm in general...

fpetzold, 2023-03-27:
Besides having no luck using my SpaceMouse with the ARM weekly build, I wonder why the latest is more than 3 months old, and from last year. I have also tried (and failed) to compile FreeCAD using conda. Please provide detailed build instructions for creating the weekly ARM Disk image. Starting from a freshly installed macOS preferably. That is, including how exactly to install and configure conda (and any other dependencies). I would assume Xcode already installed though.

I am happy to test such instructions and verify that such a build would also support the connexion devices.

Or another helpful answer would be, how was the last ARM image built, exactly?

Thanks for any help!

JohnOCFII, 2023-03-27:
@fpetzold - Another user is making great progress in getting a build environment working for macOS arm.  I was able to follow his build instructions to get a working version of FreeCAD Arm.  He's working to better integrate with the official build process, but for now, you might want to give it a shot and offer him additional feedback.

https://github.com/oursland/FreeCAD-Build-Notes

https://forum.freecad.org/viewtopic.php?t=76777&start=20

steffenxx, 2023-06-21:
...a lot of time has passed... apparently nothing further is happening in the matter of MacOS Arm for the time being. :-(

luzpaz, 2023-06-24:
It's probably been de-prioritized with all the release stuff happening. PRs are welcome!

adrianinsaval, 2023-06-26:
I have just uploaded a new arm bundle that was compiled with the driver version that was reported as working for arm, please try it: https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-33447-2023-06-26-conda-macOS-arm-py310.dmg

steffenxx, 2023-06-26:
Quick test :-) FreeCAD starts on a Mac Studio, my 3DConnexion devices (SpaceMouse) is directly supported. Files can be opened and creating a MESH is really fast in this version. You are my hero. Thanks a lot

DanielPa, 2023-06-26:
@steffenxx did you install a specific version of the driver? On MacBook Pro a quick try did not work with my SpaceMouse Compact 😕

steffenxx, 2023-06-26:
yes, 3DWare 10.7.0. This is the last driver that worked with FreeCAD. All the following drivers have never worked.

DanielPa, 2023-06-27:
Awesome! Just works like charm ❤️ 
[Driver Archive of 3Dconnexion](https://3dconnexion.com/de/archived-drivers/?fwp_drivers_os=macos-12)

Kurdie2023, 2023-06-27:
Thank you so much, Adrian. It works great! Might there be a way, that it will also work with the current 3Dconnexion driver version? I need the current version due to some other applications...

Is there a way to install this freeCAD version "permanently". Every app start I have to got to the settings to allow it.

Thanks!!

adrianinsaval, 2023-06-28:
What version of the driver? I need a direct link for download and I can make a bundle so you can test, but other people reported that later versions of the driver didn't work, there isn't much I can do since I don't neither a mac nor a 3d mouse.

About permanent installation, I have no idea, consider asking for help on the forum or google search about it.

Kurdie2023, 2023-08-07:
Hi Adrian,
Sorry for the late answer.
I just downloaded the 0.21 version for mac M1 from the website (build 33675). Unfortunately it says „No Spaceball Present“, with both driver version (10.7.0 and current version 10.8.1).

Please find the link to the current driver version below:
https://3dconnexion.com/de/drivers/

Is your adaption not implemented into this build?

Thank you!!!

best regards,

Rouven

biogoo, 2023-08-08:
Confirmed - my 3Dconnexion device is not working on my M1 mini with the 0.21.0-33675 arm64 build, but does work with the intel build (the github variant, since the version on the download page fails with a Rosetta error on launch as per https://forum.freecad.org/viewtopic.php?t=80126)

ancher1912, 2023-08-15:
The FreeCAD_weekly-builds-33674-2023-08-01-conda-macOS-arm64-py310 in combination with 10.8.1 of 3DxWare  recognises the mouse (the configuration stuff appears in the Customise menu). However, the mouse itself is still not working...

Shed2000, 2023-08-24:
> Hi Adrian, Sorry for the late answer. I just downloaded the 0.21 version for mac M1 from the website (build 33675). Unfortunately it says „No Spaceball Present“, with both driver version (10.7.0 and [...]

> What version of the driver? I need a direct link for download and I can make a bundle so you can test, but other people reported that later versions of the driver didn't work, there isn't much I can do since I don't have neither a mac nor a 3d mouse.
> 
> About permanent installation, I have no idea, consider asking for help on the forum or google search about it.

Hi Adrian

I am running 0.21 on M2 Mac with 3Dconnexion driver 10.8.1. which reports "no spaceball present." Do you have a build that works available?[
](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-33447-2023-06-26-conda-macOS-arm-py310.dmg) seems to be a dead link now?
I am new starter on FreeCAD, which looks very impressive, unfortunately I am so used to using a space mouse on CAD packages that I find it incredibly difficult to work without one!  Many Thanks.

adrianinsaval, 2023-08-24:
The build was made with version 10.7.0 of the driver since users had reported that newer versions didn't work. Please try using that version of the driver

Shed2000, 2023-08-24:
Hi Adrian
Thanks I will revert back to 10.7.0. Could you resend a link to the build you made for this version as the one you posted seems to be dead now?  Thanks

Shed2000, 2023-08-25:
freecad-0.22.0-33981 working with 10.7.0 !!!

Kurdie2023, 2023-09-01:
Where can I get this version?
Thanks!

Shed2000, 2023-09-01:
Kurdie,
I found it in the weekly builds 2023-08-24 but I can't see it there now!

Shed2000, 2023-09-01:
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Kurdie2023, 2024-01-05:
Hi all,

is there still no solution to use the 3D Connexion Spacemouse with the current official FreeCAD built and the current Connexion Driver?

Thanks!!

steffenxx, 2024-01-13:
I opened the case back in Dec 2022, since then there is always a version that works with the SpaceMouse and then again not. In the last developer versions the mouse always worked, which is nice. However, only with an old beta driver from 3D Connexion, which they had provided as a pre-release version for BigSur. I have given up hope that an "official" version of FreeCad for Silicon Macs will ever work with a current version of a 3D Connexion driver. I suspect that none of the developers use a Mac. Please don't get me wrong, I am grateful that FreeCad exists. But the day will come when the ancient driver will no longer work with a current MacOS.

adrianinsaval, 2024-01-14:
current versions use driver version 10.7.0, does that work for you?

steffenxx, 2024-01-14:
yes, it works :-) Both with the stable version and with the latest weekly build. Many thanks for the support. Why does it not work with newer versions of 3D Connexion driver? Are they so fundamentally different?

Kurdie2023, 2024-01-15:
Hi adrian and steffen,

sorry for the wrong info. I just downloaded the current freecad version from the website and the spaceball works great with the driver 10.7.0. Very nice! If it would be now possible to use it with the current 3D connexion driver, that would be amazing!
Thank you all for your great jobs!

Best regards,
Rouven

[6 more replies]

closed 2024-01-30 by chennes