FreeCAD/FreeCAD-Bundle #147: AppImage, missing dependencies, Ubuntu

anvilvapre, 2022-12-28:


luzpaz, 2022-12-28:
@looooo have you seen this issue before ?

looooo, 2022-12-28:
@anvilvapre can you try FreeCAD 0.20.2 and see if the issue remains? In my case only libfuse was missing. But I am not sure if libfuse should be bundled

```
OS: Ubuntu 22.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Austria (de_AT)
Installed mods: 
  * Assembly4 0.12.5
  * Assembly3 0.11.3

```

anvilvapre, 2022-12-28:
`sudo apt-get install libpci3` resolved it. just wanted to report it, so you can perhaps improve the app imagine.

anvilvapre, 2022-12-28:
btw, i installed xorg on top of an ubuntu-server image. so possibly ubuntu desktop installs already include pci3 as a dependency somehow(?). i suspect some python library uses the so (Qt?). https://libpci.readthedocs.io/en/latest/installation.html.

adrianinsaval, 2023-01-17:
> In my case only libfuse was missing. But I am not sure if libfuse should be bundled

from my understanding libfuse is needed to mount the appimage in the first place so bundling it wouldn't help anyways, this was a pretty widely talked about problem for ubuntu 22.04 which stopped shipping it by default and made appimages in general not able to run by default

looooo, 2023-01-19:
> from my understanding libfuse is needed to mount the appimage in the first place so bundling it wouldn't help anyways, this was a pretty widely talked about problem for ubuntu 22.04 which stopped [...]

ah thanks for the information. It's a pity libfuse is not bundled into ubuntu 22.04 anymore. But anyways it's easy to install this library via apt...

luzpaz, 2023-03-13:
Encountered again in https://github.com/FreeCAD/FreeCAD/issues/8821

adrianinsaval, 2024-09-18:
could someone please test if this issue still exist with the latest weekly? I mean the one that just got uploaded, if you have downloaded earlier today or yesterday I expect those to fail, even if they have the same revision number as the one today.

Appimages now shouldn't have LD_LIBRARY_PATH set anymore which is likely what was causing these problems with external applications

luzpaz, 2024-10-19:
How do you test this again?

adrianinsaval, 2024-10-27:
the openscad tests passed when using the openscad appimage with our latest weekly for me so I'm going to consider this fixed, feel free to reopen if it doesn't work for someone

closed 2024-10-27 by adrianinsaval

chennes, 2024-10-27:
Did the tests actually run? The OpenSCAD tests are designed to bail out early if they can't run OpenSCAD.

adrianinsaval, 2024-10-27:
> Did the tests actually run? The OpenSCAD tests are designed to bail out early if they can't run OpenSCAD.

I don't think that's the case, they fail on flatpak where you still can't run openscad appimage. Anyways I ran it again and it seems to have run it:
```
16:57:38  Creating /tmp/tmpl0j6wg8r/complex-fuse.scad
16:57:38  WARNING: Token 'DOT' defined, but not used
16:57:38  WARNING: Token 'WORD' defined, but not used
16:57:38  WARNING: There are 2 unused tokens
16:57:38  obj [<Part::PartFeature>] has no Name & Shape
16:57:38  obj [<Part::PartFeature>] has no Name & Shape
16:57:38  <Exception> Input shape is null
16:57:38  Recompute failed : union
16:57:38  <Exception> Input shape is null
16:57:38  Recompute failed : union
16:57:38  End processing CSG file
16:57:39  Creating /tmp/tmpl0j6wg8r/empty_union.scad
16:57:39  WARNING: Cylinder parameters ambiguous in file empty_union.scad, line 53
WARNING: Cylinder parameters ambiguous in file empty_union.scad, line 57

16:57:39  WARNING: Token 'DOT' defined, but not used
16:57:39  WARNING: Token 'WORD' defined, but not used
16:57:39  WARNING: There are 2 unused tokens
16:57:39  End processing CSG file
16:57:40  Creating /tmp/tmpl0j6wg8r/circle.scad
16:57:40  WARNING: Token 'DOT' defined, but not used
16:57:40  WARNING: Token 'WORD' defined, but not used
16:57:40  WARNING: There are 2 unused tokens
16:57:40  End processing CSG file
16:57:41  Creating /tmp/tmpl0j6wg8r/cube.scad
16:57:41  WARNING: Token 'DOT' defined, but not used
16:57:41  WARNING: Token 'WORD' defined, but not used
16:57:41  WARNING: There are 2 unused tokens
16:57:41  End processing CSG file
16:57:41  Creating /tmp/tmpl0j6wg8r/cylinder.scad
16:57:41  WARNING: Token 'DOT' defined, but not used
16:57:41  WARNING: Token 'WORD' defined, but not used
16:57:41  WARNING: There are 2 unused tokens
16:57:41  End processing CSG file
16:57:42  Creating /tmp/tmpl0j6wg8r/difference.scad
16:57:42  WARNING: Token 'DOT' defined, but not used
16:57:42  WARNING: Token 'WORD' defined, but not used
16:57:42  WARNING: There are 2 unused tokens
16:57:42  End processing CSG file
16:57:42  Creating /tmp/tmpl0j6wg8r/import_stl.scad
16:57:42  WARNING: Token 'DOT' defined, but not used
16:57:42  WARNING: Token 'WORD' defined, but not used
16:57:42  WARNING: There are 2 unused tokens
16:57:42  End processing CSG file
16:57:43  Creating /tmp/tmpl0j6wg8r/intersection.scad
16:57:43  WARNING: Token 'DOT' defined, but not used
16:57:43  WARNING: Token 'WORD' defined, but not used
16:57:43  WARNING: There are 2 unused tokens
16:57:43  End processing CSG file
16:57:45  Creating /tmp/tmpl0j6wg8r/linear_extrude_simple.scad
16:57:45  WARNING: Token 'DOT' defined, but not used
16:57:45  WARNING: Token 'WORD' defined, but not used
16:57:45  WARNING: There are 2 unused tokens
16:57:45  End processing CSG file
16:57:45  Creating /tmp/tmpl0j6wg8r/linear_extrude_twist.scad
16:57:45  WARNING: Token 'DOT' defined, but not used
16:57:45  WARNING: Token 'WORD' defined, but not used
16:57:45  WARNING: There are 2 unused tokens
16:57:45  End processing CSG file
16:57:45  Creating /tmp/tmpl0j6wg8r/linear_extrude_scale.scad
16:57:45  WARNING: Token 'DOT' defined, but not used
16:57:45  WARNING: Token 'WORD' defined, but not used
16:57:45  WARNING: There are 2 unused tokens
16:57:45  End processing CSG file
16:57:45  Creating /tmp/tmpl0j6wg8r/linear_extrude_twist_scale.scad
16:57:45  WARNING: Token 'DOT' defined, but not used
[... 110 lines cut ...]
16:57:58  WARNING: Token 'WORD' defined, but not used
16:57:58  WARNING: There are 2 unused tokens
16:57:58  End processing CSG file
16:57:59  Creating /tmp/tmpl0j6wg8r/union.scad
16:57:59  WARNING: Token 'DOT' defined, but not used
16:57:59  WARNING: Token 'WORD' defined, but not used
16:57:59  WARNING: There are 2 unused tokens
16:57:59  End processing CSG file
16:57:59  WARNING: Token 'DOT' defined, but not used
16:57:59  WARNING: Token 'WORD' defined, but not used
16:57:59  WARNING: There are 2 unused tokens
16:57:59  End processing CSG file
16:58:00  ECHO: version = [2021, 1, 0]

16:58:00  WARNING: Token 'DOT' defined, but not used
16:58:00  WARNING: Token 'WORD' defined, but not used
16:58:00  WARNING: There are 2 unused tokens
16:58:00  End processing CSG file
16:58:01  
----------------------------------------------------------------------
Ran 26 tests in 23.9s

16:58:01  OK
```
I had system monitor open and I saw the openscad appimage running many times

adrianinsaval, 2024-10-27:
for comparison flatpak fails like this:
```
Creating /var/tmp/tmpjayrr5th/complex-fuse.scad
======================================================================
ERROR: test_complex_fuse_no_placement (OpenSCADTest.app.test_importCSG.TestImportCSG.test_complex_fuse_no_placement)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.11/unittest/case.py", line 57, in testPartExecutor
    yield
  File "/usr/lib/python3.11/unittest/case.py", line 623, in run
    self._callTestMethod(testMethod)
  File "/usr/lib/python3.11/unittest/case.py", line 579, in _callTestMethod
    if method() is not None:
       ^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADTest/app/test_importCSG.py", line 546, in test_complex_fuse_no_placement
    doc = self.utility_create_scad(csg_data, "complex-fuse")
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADTest/app/test_importCSG.py", line 82, in utility_create_scad
    return importCSG.open(filename)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/importCSG.py", line 118, in open
    tmpfile = callopenscad(filename)
              ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 169, in callopenscad
    return call_openscad_with_pipes(inputfilename, outputfilename, outputext, keepname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 214, in call_openscad_with_pipes
    raise OpenSCADError('%s %s\n' % (stdoutd.strip(),stderrd.strip()))
OpenSCADUtils.OpenSCADError: ' dlopen(): error loading libfuse.so.2\n\nAppImages require FUSE to run. \nYou might still be able to extract the contents of this AppImage \nif you run it with the --appimage-extract option. \nSee https://github.com/AppImage/AppImageKit/wiki/FUSE \nfor more information\n'

Creating /var/tmp/tmpjayrr5th/empty_union.scad
======================================================================
ERROR: test_empty_union (OpenSCADTest.app.test_importCSG.TestImportCSG.test_empty_union)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.11/unittest/case.py", line 57, in testPartExecutor
    yield
  File "/usr/lib/python3.11/unittest/case.py", line 623, in run
    self._callTestMethod(testMethod)
  File "/usr/lib/python3.11/unittest/case.py", line 579, in _callTestMethod
    if method() is not None:
       ^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADTest/app/test_importCSG.py", line 513, in test_empty_union
    doc = self.utility_create_scad(content, "empty_union")
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADTest/app/test_importCSG.py", line 82, in utility_create_scad
    return importCSG.open(filename)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/importCSG.py", line 118, in open
    tmpfile = callopenscad(filename)
              ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 169, in callopenscad
    return call_openscad_with_pipes(inputfilename, outputfilename, outputext, keepname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 214, in call_openscad_with_pipes
    raise OpenSCADError('%s %s\n' % (stdoutd.strip(),stderrd.strip()))
OpenSCADUtils.OpenSCADError: ' dlopen(): error loading libfuse.so.2\n\nAppImages require FUSE to run. \nYou might still be able to extract the contents of this AppImage \nif you run it with the --appimage-extract option. \nSee https://github.com/AppImage/AppImageKit/wiki/FUSE \nfor more information\n'

Creating /var/tmp/tmpjayrr5th/circle.scad
======================================================================
ERROR: test_import_circle (OpenSCADTest.app.test_importCSG.TestImportCSG.test_import_circle)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.11/unittest/case.py", line 57, in testPartExecutor
    yield
  File "/usr/lib/python3.11/unittest/case.py", line 623, in run
    self._callTestMethod(testMethod)
  File "/usr/lib/python3.11/unittest/case.py", line 579, in _callTestMethod
    if method() is not None:
       ^^^^^^^^
[... 423 lines cut ...]
    yield
  File "/usr/lib/python3.11/unittest/case.py", line 623, in run
    self._callTestMethod(testMethod)
  File "/usr/lib/python3.11/unittest/case.py", line 579, in _callTestMethod
    if method() is not None:
       ^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADTest/app/test_importCSG.py", line 48, in test_open_scad
    doc = importCSG.open(testfile)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/importCSG.py", line 118, in open
    tmpfile = callopenscad(filename)
              ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 169, in callopenscad
    return call_openscad_with_pipes(inputfilename, outputfilename, outputext, keepname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 214, in call_openscad_with_pipes
    raise OpenSCADError('%s %s\n' % (stdoutd.strip(),stderrd.strip()))
OpenSCADUtils.OpenSCADError: ' dlopen(): error loading libfuse.so.2\n\nAppImages require FUSE to run. \nYou might still be able to extract the contents of this AppImage \nif you run it with the --appimage-extract option. \nSee https://github.com/AppImage/AppImageKit/wiki/FUSE \nfor more information\n'

----------------------------------------------------------------------
Ran 26 tests in 3.09s

FAILED (errors=18)

```