FreeCAD/FreeCAD-Bundle #150: [Issue] [path] no shapes in favorite folder after fresh installation of FreeCAD Appimage

walterwissmann, 2023-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=75237

### Version

0.21 (Development)

### Full version info

```shell
OS: Linux Mint 21.1 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31556 (Git) AppImage
Build type: Release
Branch: master
Hash: 4091db47397f6e537a284f50c4f572545f3f30d0
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * kicadStepUpMod 10.16.8
  * ExplodedAssembly
  * TabBar
  * A2plus 0.4.60h
  * Lithophane
  * DynamicData 2.46.0
  * fasteners 0.4.52
```

### Subproject(s) affected?

None

### Issue description

1. Rename /home/user/.config/FreeCAD to became a near fresh installation
2. Start FreeCAD
3. Switch to PATH WB
4. Click on  **ToolBit Dock** to open the procedure to create a favorite folder and copy examples to them.

5. The directory **Shape** was created but nothing copied to them.
6. At the **Parameter Editor** an there the **LastPathToolShape** at /Preferences/Mod/Path was not generated

7. It was generated when i define a new toolbit, but with the link to the TMP folder.

For a better illustration look at the thread in FreeCAD Forum find in the link above.

Thanks to all of you.

Walter

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-14:
Moved to https://github.com/FreeCAD/FreeCAD-Bundle/issues/150 since this is AppImage related

walterwissmann, 2023-01-14:
Hey @luzpaz 

It was clear to me that the problem was a path problem. How do you come to the conclusion that it would be an appimage problem?

Cheers
Walter

luzpaz, 2023-01-14:
@walterwissmann out of curiosity, can you reproduce this on Snap edge ? https://github.com/FreeCAD/FreeCAD-snap

walterwissmann, 2023-01-16:
Ok...

I see i need help @luzpaz. 

I think I need the snapd to install FreeCAD-Snap. But I can't install the snap, see:

[image]

[image]

Do you know how to install snapd?

Walter

walterwissmann, 2023-01-16:
> @walterwissmann out of curiosity, can you reproduce this on Snap edge ? https://github.com/FreeCAD/FreeCAD-snap

Now i installed the "snap edge" of FreeCad.

And it is the same problem on: 

OS: Ubuntu Core 20 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31601 (Git) Snap 549
Build type: Release
Branch: master
Hash: 32bbdb0ab25e6c04c03d5e793b43c36a9fe94164
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: German/Germany (de_DE)

Walter