FreeCAD/FreeCAD-Bundle #151: hardcoded path in create_bundle.bat

adrianinsaval, 2023-01-17:
I don't know what these are supposed to do but I highly doubt they are working as this path wouldn't exist when running the action
https://github.com/FreeCAD/FreeCAD-Bundle/blob/5720f1fca062579e198554629981f255ea9e3f45/conda/win_py38/create_bundle.bat#L64
https://github.com/FreeCAD/FreeCAD-Bundle/blob/5720f1fca062579e198554629981f255ea9e3f45/conda/win_py310/create_bundle.bat#L64

looking at the actions logs there always seem to be this line:

> The system cannot find the path specified.

I suspect it's due to this, it's confusing and time wasting when trying to troubleshoot (I'm trying to figure out why windows bundle uploads are failing)

luzpaz, 2023-03-29:
@adrianinsaval did you eventually figure out what these hardcoded paths were about?

luzpaz, 2023-03-30:
@looooo ping

closed 2023-11-14 by adrianinsaval