FreeCAD/FreeCAD-Bundle #155: AppImage icon is still missing on Desktop Integration (Ubuntu 22.10)

leoheck, 2023-02-23:
Hi guys, 

The desktop icon is still missing on my system.
This problem was never fixed. 
Do you guys have any idea what I can do to fix this?

[image]

I have a single Desktop launcher (which is good so I am not making any mistakes on this)

[image]

The latest entry of the icon looks like this one

[image]

I found that there are also tons of freecad icons available but none of them are being used.
These may be leftovers of the app image update tool. (so maybe their issue)

[image]

The referenced icon exists in the `.local/share/application/icons/default/` directory

[image]

The image of the icon is good too.

[image]

I have no clue what is the issue. 
Maybe the desktop file does not look for things inside the  `.local/share/application/icons/default/`.

I am using `Ubuntu 22.10`
I am using the weekly AppImages 
The current one is `FreeCAD_weekly-builds-31917-2023-02-23-conda-Linux-x86_64-py310.AppImage`
I update Freecad Weekly very often.

This is my old and related one. 
https://github.com/FreeCAD/FreeCAD-Bundle/issues/140

Freecad version by the way.

```
OS: Ubuntu 22.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git) AppImage
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.0
  * Curves 0.6.6
  * fasteners 0.4.54
  * lattice2 1.0.0
  * FreeCAD_Assembly4 0.12.5
  * kicadStepUpMod 10.17.1
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
```

leoheck, 2023-02-23:
Taking as an example another AppImage that has this icon working is appimagetool.

[image]

```
Icon=appimagekit_af764131df8423acdc4dd04f5fc240eb_appimagetool
```

I found those icons in a different place:

[image]

<hr/>

Another example is Arduino

[image]

```
Icon=appimagekit_d8c00426ed8039b90e3b484cfd18f3f6_arduino-ide
```

The icon is here

[image]

<hr/>

After placing the Freecad icon manually in the above folders... it works! Good.

[image]

```
../icons/hicolor/64x64/apps/appimagekit_15cb8d728deb6ae8dedec56eb40ab4e9_freecad_weekly.png
../icons/hicolor/128x128/apps/appimagekit_15cb8d728deb6ae8dedec56eb40ab4e9_freecad_weekly.png
../icons/hicolor/256x256/apps/appimagekit_15cb8d728deb6ae8dedec56eb40ab4e9_freecad_weekly.png
```

Just another issue.
The existing Freecad Icon is 60x60... this one was resized to 64x64 to be placed inside the 64x64 folder
The others were resized to 128x128 and 256x256 respectively. 

It would be nice to have this change in the Freecad AppImage.
And also the right resolution for each image, of course.

adrianinsaval, 2023-03-22:
I don't think any of this is handled by this appimage so there is nothing to do here, this is a problem in your system and whatever tool you are using for integrating your appimages.

luzpaz, 2023-03-22:
Maybe checking upstream with the AppImage folks ?

leoheck, 2023-03-22:
> this is a problem in your system and whatever tool you are using for integrating your appimages.

@adrianinsaval hmm perhaps this is the issue. I am using Ubuntu 22.10 with Gnome, with AppImage Launcher. 

Now that you mentioned it, I thought the AppImage Launcher was the right way to use these things. But it may not be the standard/official or the only method, indeed. 

How do you manage AppImages? Do you have a recommendation for me to try?

leoheck, 2023-03-22:
> Maybe checking upstream with the AppImage folks ?

@luzpaz, I did not understand what you are saying, could you explain it better?

adrianinsaval, 2023-03-22:
I used appimagelauncher at some point but then got tired of it's shenanigans and uninstalled it, I kept the FreeCAD desktop file and edited it myself, changed the icon to something persistent and also renamed the appimage to something persistent (FreeCAD.AppInage) and use a script to auto rename the appimage to this when updating.

leoheck, 2023-03-22:
Yeah, I am doing this, but we could have this working. But I like to keep Freecad Weekly updated to taste the latest changes. And fixing this, somehow, could help me, you, and others. I will investigate with AppLauncher is we should be doing something different here or not.

leoheck, 2023-03-22:
I found this appimagelint, I am not sure if we have to follow it or how good this is, but at least it gives us a hint on where the issues may be. The following picture is just the end of the command, that is relevant here but I am attaching the output as text too.

[image]

```bash
~/Applications …
➜ ./appimagelint-x86_64.AppImage FreeCAD_weekly-builds-31917-2023-03-19-conda-Linux-x86_64-py310.AppImage 
appimagelint.cli[597535] [INFO] Checking AppImage FreeCAD_weekly-builds-31917-2023-03-19-conda-Linux-x86_64-py310.AppImage
appimagelint.cli[597535] [INFO] Running check "GNU libc ABI check"
appimagelint.glibc_abi_check[597535] [INFO] detected required version for runtime: 2.14
appimagelint.glibc_abi_check[597535] [INFO] detected required version for payload: 2.33
appimagelint.cache[597535] [INFO] Fetching glibc version data for debian
appimagelint.setup[597535] [INFO] Fetching glibc package versions from Debian sources API
appimagelint.glibc_abi_check[597535] [INFO] [✖] AppImage can run on Debian oldstable (buster)
appimagelint.glibc_abi_check[597535] [INFO] [✖] AppImage can run on Debian stable (bullseye)
appimagelint.glibc_abi_check[597535] [INFO] [✔] AppImage can run on Debian testing (bookworm)
appimagelint.glibc_abi_check[597535] [INFO] [✔] AppImage can run on Debian unstable (sid)
appimagelint.cache[597535] [INFO] Fetching glibc version data for ubuntu
appimagelint.setup[597535] [INFO] Fetching glibc package versions from Ubuntu FTP mirror
appimagelint.glibc_abi_check[597535] [INFO] [✔] AppImage can run on Ubuntu kinetic
appimagelint.glibc_abi_check[597535] [INFO] [✔] AppImage can run on Ubuntu jammy
appimagelint.glibc_abi_check[597535] [INFO] [✖] AppImage can run on Ubuntu focal
appimagelint.glibc_abi_check[597535] [INFO] [✖] AppImage can run on Ubuntu bionic
appimagelint.glibc_abi_check[597535] [INFO] [✖] AppImage can run on Ubuntu xenial
appimagelint.glibc_abi_check[597535] [INFO] [✖] AppImage can run on Ubuntu trusty
appimagelint.cache[597535] [INFO] Fetching glibc version data for centos
appimagelint.setup[597535] [INFO] Fetching glibc package versions from CentOS mirror
appimagelint.glibc_abi_check[597535] [INFO] [✖] AppImage can run on CentOS 7
appimagelint.cli[597535] [INFO] Running check "GNU libstdc++ ABI check"
appimagelint.glibcxx_abi_check[597535] [INFO] detected required version for runtime: <none>
appimagelint.glibcxx_abi_check[597535] [INFO] detected required version for payload: 3.4.30
appimagelint.cache[597535] [INFO] Fetching glibcxx version data for debian
appimagelint.gnu_lib_versions_symbols_finder[597535] [WARNING] no binaries found in /dev/shm/appimagelint-5gvg3arn.tmp/out/
appimagelint.gnu_lib_versions_symbols_finder[597535] [WARNING] no binaries found in /dev/shm/appimagelint-fndju916.tmp/out/
appimagelint.gnu_lib_versions_symbols_finder[597535] [WARNING] no binaries found in /dev/shm/appimagelint-pugd44a3.tmp/out/
appimagelint.gnu_lib_versions_symbols_finder[597535] [WARNING] no binaries found in /dev/shm/appimagelint-0yafo_a4.tmp/out/
appimagelint.glibcxx_abi_check[597535] [INFO] [✖] AppImage can run on Debian oldstable (buster)
appimagelint.glibcxx_abi_check[597535] [INFO] [✖] AppImage can run on Debian stable (bullseye)
appimagelint.glibcxx_abi_check[597535] [INFO] [✖] AppImage can run on Debian testing (bookworm)
appimagelint.glibcxx_abi_check[597535] [INFO] [✖] AppImage can run on Debian unstable (sid)
appimagelint.cache[597535] [INFO] Fetching glibcxx version data for ubuntu
appimagelint.gnu_lib_versions_symbols_finder[597535] [WARNING] no binaries found in /dev/shm/appimagelint-n6u_x6xf.tmp/out/
appimagelint.gnu_lib_versions_symbols_finder[597535] [WARNING] no binaries found in /dev/shm/appimagelint-ejazizt8.tmp/out/
appimagelint.gnu_lib_versions_symbols_finder[597535] [WARNING] no binaries found in /dev/shm/appimagelint-qrigh4ny.tmp/out/
appimagelint.gnu_lib_versions_symbols_finder[597535] [WARNING] no binaries found in /dev/shm/appimagelint-vqqpktak.tmp/out/
appimagelint.gnu_lib_versions_symbols_finder[597535] [WARNING] no binaries found in /dev/shm/appimagelint-bwriaghg.tmp/out/
appimagelint.gnu_lib_versions_symbols_finder[597535] [WARNING] no binaries found in /dev/shm/appimagelint-tan80359.tmp/out/
appimagelint.glibcxx_abi_check[597535] [INFO] [✔] AppImage can run on Ubuntu kinetic
appimagelint.glibcxx_abi_check[597535] [INFO] [✔] AppImage can run on Ubuntu jammy
appimagelint.glibcxx_abi_check[597535] [INFO] [✖] AppImage can run on Ubuntu focal
appimagelint.glibcxx_abi_check[597535] [INFO] [✖] AppImage can run on Ubuntu bionic
appimagelint.glibcxx_abi_check[597535] [INFO] [✖] AppImage can run on Ubuntu xenial
appimagelint.glibcxx_abi_check[597535] [INFO] [✖] AppImage can run on Ubuntu trusty
appimagelint.cache[597535] [INFO] Fetching glibcxx version data for centos
appimagelint.setup[597535] [INFO] Fetching libstdc++ package versions from CentOS mirror
appimagelint.glibcxx_abi_check[597535] [INFO] [✖] AppImage can run on CentOS 7
appimagelint.cli[597535] [INFO] Running check "Icons validity and location check"
appimagelint.icons_check[597535] [INFO] Extracting icon name from desktop file: /tmp/.mount_FreeCAk4Jhzv/freecad_weekly.desktop
appimagelint.icons_check[597535] [WARNING] Multiple matching icons found in AppDir root, checking all
appimagelint.icons_check[597535] [INFO] Checking resolution of icon: /tmp/.mount_FreeCAk4Jhzv/freecad_weekly.png
[... cut ...] ce in the application menu
appimagelint.desktop_files[597535] [INFO] Checking desktop file /tmp/.mount_FreeCAvkdizH/freecad_weekly.desktop with desktop-file-validate
/tmp/.mount_FreeCAvkdizH/freecad_weekly.desktop: hint: value "Engineering;" for key "Categories" in group "Desktop Entry" does not contain a registered main category; application might only show up in a "catch-all" section of the application menu
appimagelint.desktop_files[597535] [INFO] [✔] All desktop files in AppDir are valid
```

TheAssassin, 2023-03-22:
The log in https://github.com/FreeCAD/FreeCAD-Bundle/issues/155#issuecomment-1480158874 hints that the most likely issue with the icons is that it has an incorrect resolution. Desktop environments are very picky about this.

I integrated the AppImage using `ail-cli integrate` (shouldn't really matter, though; in the end, most tools just use https://github.com/AppImage/libappimage, so they work the same). Log output excerpt:

```
WARNING: Unable to resize the application icon into a 128x128 image: "Unable to load image.". It will be written unchanged.
WARNING: Unable to resize the application icon into a 256x256 image: "Unable to load image.". It will be written unchanged.
```

Not sure why this happens, but that's another issue. This is, after all, just an attempt to make a broken icon work nevertheless. It's always better not to rely on this logic.

I cleaned up the icon mess as follows:
- resize the 60x60 icon in the root directory to 64x64 (honestly, there's an SVG, so we could just dump it, too)
- remove the `usr/share/icons/default` directory tree including the contents
- move `freecad_weekly.svg` into `usr/share/icons/hicolor/scalable/apps` (an XDG icon theme spec compliant directory path)
- move the now 64x64 `freecad_weekly.png` into `usr/share/icons/hicolor/64x64/apps` (an XDG icon theme spec compliant directory path, too)

Et voilà, working result:

[image]

It is documented (somewhere) that the integration algorithm just copies the icons following their layout in `usr/share/icons`. If these paths violate the XDG icon theme spec or the contents are invalid due to, e.g., invalid icon sizes, they will be ignored by virtually any desktop implementing this specification. It's really that simple. The renaming is done to avoid collisions when having multiple AppImages integrated, but that doesn't have any influence on the result.

We see that appimagelint's report is correct and provides the right hints to fix the problem and provide a compliant outcome. No suitable icon was found, as it reports:

```
appimagelint.icons_check[597535] [ERROR] Could not extract resolution from icon path,should be usr/share/icons/<theme>/<res>/apps/<name>.<ext>
```

@leoheck all these old icons are just leftovers. Feel free to clean them up. I'm not sure if AppImageLauncher's built-in updater does that; however, appimagelauncherd should see that the old AppImage is gone and clean up leftovers. In any case, these are at most a few MiB, so nothing to worry about, really.

leoheck, 2023-03-23:
Thanks for the backup @TheAssassin.
I applied those changes to the PR listed above.

closed 2023-03-23 by chennes

leoheck, 2023-03-23:
@TheAssassin @adrianinsaval 

It worked flawlessly. 
The one on the left is my old low-quality workaround. The one in the right is the new and pretty 64x64 icon.

[image]

@TheAssassin thank you, this botters me so much for such a long time.

leoheck, 2023-03-24:
@adrianinsaval the last weekly had the icon working, but the AppImage was not working.

Is there a way to get the previously working AppImage?

See:
```
~ took 58m 1.7s …
➜ freecad               
Latest FreeCAD /home/lheck/Applications/FreeCAD_weekly-builds-32457-2023-03-23-conda-Linux-x86_64-py310_19a8ec00bff997b4f6aeb41fc2387015.AppImage
FreeCAD 0.21.0, Libs: 0.21.0R32457 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench (v0.12.6) loaded
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
/tmp/.mount_FreeCAdBikK5/AppRun: line 43: 523140 Aborted                 (core dumped) ${MAIN} "$@"
```