FreeCAD/FreeCAD-Bundle #161: Qt WebEngine errors on Windows 11

erickwill, 2023-03-26:
From a few releases, almost a month ago, I started to have these issues: 

Qt WebEngine locales directory not found at location C:/Users/erick/Documents/Apps/FreeCAD_weekly-builds-32532-2023-03-26-conda-Windows-x86_64-py310/bin\qtwebengine_locales. Trying fallback directory... Translations MAY NOT not be correct.

Installed Qt WebEngine locales directory not found at location C:/Users/erick/Documents/Apps/FreeCAD_weekly-builds-32532-2023-03-26-conda-Windows-x86_64-py310/translations\qtwebengine_locales. Trying application directory...

Any fix or workaround?
I'm running the FreeCAD_weekly-builds-32532-2023-03-26-conda-Windows-x86_64-py310 under an up-to-dated Windows 11.

chennes, 2023-03-26:
Should be fixed by #162.

adrianinsaval, 2023-04-12:
since this is fixed, I'm closing this issue

closed 2023-04-12 by adrianinsaval