FreeCAD/FreeCAD-Bundle #165: Appimage unable to install addons via addon-manager

einhander, 2023-04-04:
OS: Debian 12

````
FreeCAD 0.20.2, Libs: 0.20.2R29603 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench (v0.12.6) loaded
Sheet Metal workbench loaded
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: Нет такого файла или каталога
AddonManager: Unable to fetch git updates for workbench 3DfindIT

Cmd('git') failed due to: exit code(128)

  cmdline: git fetch
  stderr: '/usr/lib/git-core/git-remote-https: symbol lookup error: /lib/x86_64-linux-gnu/libcurl-gnutls.so.4: undefined symbol: nghttp2_option_set_no_rfc9113_leading_and_trailing_ws_validation'
AddonManager: Unable to fetch git updates for workbench A2plus
Cmd('git') failed due to: exit code(128)
  cmdline: git fetch
  stderr: '/usr/lib/git-core/git-remote-https: symbol lookup error: /lib/x86_64-linux-gnu/libcurl-gnutls.so.4: undefined symbol: nghttp2_option_set_no_rfc9113_leading_and_trailing_ws_validation'
AddonManager: Unable to fetch git updates for workbench Assembly4
Cmd('git') failed due to: exit code(128)
  cmdline: git fetch
  stderr: '/usr/lib/git-core/git-remote-https: symbol lookup error: /lib/x86_64-linux-gnu/libcurl-gnutls.so.4: undefined symbol: nghttp2_option_set_no_rfc9113_leading_and_trailing_ws_validation'
AddonManager: Unable to fetch git updates for workbench BIM
Cmd('git') failed due to: exit code(128)
  cmdline: git fetch
  stderr: '/usr/lib/git-core/git-remote-https: symbol lookup error: /lib/x86_64-linux-gnu/libcurl-gnutls.so.4: undefined symbol: nghttp2_option_set_no_rfc9113_leading_and_trailing_ws_validation'
AddonManager: Unable to fetch git updates for workbench CfdOF
Cmd('git') failed due to: exit code(128)
  cmdline: git fetch
  stderr: '/usr/lib/git-core/git-remote-https: symbol lookup error: /lib/x86_64-linux-gnu/libcurl-gnutls.so.4: undefined symbol: nghttp2_option_set_no_rfc9113_leading_and_trailing_ws_validation'
````

FreeCad from OS repository can install addons without errors

chennes, 2023-04-04:
Can you try disabling git in the Addon Manager preferences?

adrianinsaval, 2023-05-04:
https://www.reddit.com/r/Ubuntu/comments/134x93o/git_clone_not_working/

einhander, 2023-05-06:
conda: command not found

chennes, 2023-05-06:
It's likely that AppImage users will not be able to use the Git access methods: can you try disabling Git in the Addon Manager preferences?

einhander, 2023-05-07:
> The same error: `Cmd('git') failed due to: exit code(128)`

@chennes Yes, it's the same error. It's looks like some library versions mismatch. Git from appimage worked a months ago.

chennes, 2023-05-07:
Oh, I see, I missed that this was 0.20.

adrianinsaval, 2023-07-15:
is this happening on the 0.21 bundle too?

adrianinsaval, 2023-08-17:
no response, not actionable -> closing

Please reopen if the issue is still present on the 0.21 or 0.22 appimage

closed 2023-08-17 by adrianinsaval