FreeCAD/FreeCAD-Bundle #166: [Bug]  MacOS crash at startup all the time (critical)

krl91, 2023-04-06:


luzpaz, 2023-04-06:
Moved to [FreeCAD-Bundle](https://github.com/FreeCAD/FreeCAD-Bundle/)

adrianinsaval, 2023-04-12:
looo mentioned 2a602fbea9479434e84ab66f40bd601518ef9622 as potential fix, but I see it was only applied to osx-arm script, maybe it should be applied to the regular osx bundle script too? I suggest one of the interested parties to make a PR doing this.

chennes, 2023-04-12:
If my quick scan of this topic is correct, then https://github.com/FreeCAD/FreeCAD-Bundle/commit/7dc3210c11cf2f27317f30c289838a0d998c3298 will fix this

adrianinsaval, 2023-04-12:
When a commit is made to this repo a new bundle is made automatically, a fixed bundle should be available in ~30min

adrianinsaval, 2023-04-12:
the new bundle is available, please test and report.

luzpaz, 2023-04-13:
@krl91 :point_up_2:

krl91, 2023-04-13:
Hi,
I just tested with https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-32532-2023-04-12-conda-macOS-x86_64-py310.dmg
- First problem: the name of the folder is still the same
- Same crash at start up
Can you rename this folder (on macOS "Resources" folder name is standard)?  why is it so complicated?

ls /Applications/FreeCAD.app/Contents/Resources/
Ext			freecad-doc.icns	qml
LICENSE			freecad-script.icns	qt.conf
Mod			freecad.icns		**resources**
bin			lib			sbin
cmake			libexec			share
conda-meta		man			ssl
doc			mkspecs			translations
etc			phrasebooks		var
fonts			plugins

also the solution on https://github.com/FreeCAD/FreeCAD-Bundle/commit/7dc3210c11cf2f27317f30c289838a0d998c3298 is a quick and dirty one: why creating a link for a folder that has the wrong name...

@luzpaz

adrianinsaval, 2023-11-14:
870a89bd64c8e0183fe8e8c7c111a41ac3d07785 should have fixed this, I'm going to assum it did. Please comment or reopen if this issue is still present on the latest release

closed 2023-11-14 by adrianinsaval