FreeCAD/FreeCAD-Bundle #168: Weekly MacOS ARM build requires Rosetta

alber70g, 2023-04-14:
[image]

JohnOCFII, 2023-04-14:
This recently updated process allows me to build and run FreeCAD on my M1 arm Mac.  My understanding is that this was going to be integrated into the automated build process for weekly builds, because as you can see from the date above -- that arm build is quite old.  

https://github.com/oursland/FreeCAD-Build-Notes

I'm not sure what remains before the steps in that process can be integrated into the weekly automated build process.

luzpaz, 2023-04-16:
Wow, excellent effort in https://github.com/oursland/FreeCAD-Build-Notes !

> I'm not sure what remains before the steps in that process can be integrated into the weekly automated build process.

How do you suggest we ascertain that?

realthunder, 2023-04-16:
I happen to be working on github auto build recently. I managed to make it work to [cross compile](https://github.com/realthunder/FreeCAD/blob/LinkMerge/.github/workflows/conda.yml) osx arm conda package on github intel machine. But to my understanding, we still need a real osx arm machine to make the bundle, which github does not offer at the moment. There is only commercial alternatives, which are not cheap. I am curious to see if anyone has any way around it. Or, maybe find some one to volunteer his personal osx arm machine as a [self-hosted](https://docs.github.com/en/actions/hosting-your-own-runners/about-self-hosted-runners) github runner.

luzpaz, 2023-04-16:
> I am curious to see if anyone has any way around it. Or, maybe find some one to volunteer his personal osx arm machine as a [self-hosted](https://docs.github.com/en/actions/hosting-your-own-runners/about-self-hosted-runners) github runner.

Sound something that the FPA should be involved in ?

JohnOCFII, 2023-04-16:
> How do you suggest we ascertain that?

@oursland may have some ideas.

chennes, 2023-04-16:
I have an M1 that we can use as a self-hosted runner, someone just needs to tell me what I need to do to set such a thing up.

luzpaz, 2023-04-16:
@realthunder looks like chennes is game :tada:

alber70g, 2023-05-04:
Is it possible to add a new ARM64 build for MacOS to either weekly or 0.20.2 (https://github.com/FreeCAD/FreeCAD-Bundle/releases)?

adrianinsaval, 2023-05-04:
can we emulate arm on a github action?

chennes, 2023-05-04:
We don't need any emulation, I just need to set up a self-hosted runner -- I have a lot of balls in the air right now, though, so it hasn't made my priority list.

adrianinsaval, 2023-05-04:
can you run linux and windows vms on it too?

chennes, 2023-05-04:
Yes, but I haven't tried compiling on Windows because Visual Studio doesn't support running on ARM (you are supposed to cross-compile from x86).

adrianinsaval, 2023-05-05:
> Yes, but I haven't tried compiling on Windows because Visual Studio doesn't support running on ARM (you are supposed to cross-compile from x86).

that is so stupid but I don't know what I was expecting from MS

> We don't need any emulation, I just need to set up a self-hosted runner -- I have a lot of balls in the air right now, though, so it hasn't made my priority list.

I think there is a runner available in azure to get mac arm builds but it keeps failing, we could use emulation for the bundling process that happens here on github

adrianinsaval, 2023-06-26:
I have uploaded a new arm bundle, can you try if it also asks for rosetta? https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-33447-2023-06-26-conda-macOS-arm-py310.dmg

JohnOCFII, 2023-06-26:
> I have uploaded a new arm bundle, can you try if it also asks for rosetta? https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-33447-2023-06-26-conda-macOS-arm-py310.dmg

Seems to work fine on my M1 MacBook Pro.  Activity Viewer shows it at an Apple (arm) binary. I saw no request for Rosetta.  FYI - It only shows the generic App icon.  
[image]

```
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33447 (Git)
Build type: Release
Branch: master
Hash: 090ae4571a589171dccb5dcabaf9d71e453ab2f3
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.7.1
Locale: C/Default (C)
```

adrianinsaval, 2023-06-26:
Good, probably the other bundle was at least partially cross compiled and that's why it asks for Rosetta, I had this version compiled using cirrus-ci's M1 runners and then bundled in chennes' M1 machine so it should be all native.
I don't know about the icon, would have to look into how that's supposed to be specified but would love it if any of the Mac connoisseurs could lend me a hand there

closed 2023-06-26 by adrianinsaval

JohnOCFII, 2023-06-26:
> I don't know about the icon, would have to look into how that's supposed to be specified but would love it if any of the Mac connoisseurs could lend me a hand there

Looking in the package contents, comparing it to a recent Intel Mac build, there are 3 `icns` files that need to be in the Resources directory.  These are currently (properly) referenced from the Info.plist file.  I copied over the 3 files, but that alone didn't seem to make the icon show up, so we'll need some one with real Mac developer background to explain how to hook them up properly.

[image]