FreeCAD/FreeCAD-Bundle #170: [Issue] Program received signal SIGSEGV at startup (Debian testing, python 3.11.1, Qt 5.15.7)

marcogiusti, 2023-04-20:


adrianinsaval, 2023-05-04:
@luzpaz if I understood correctly this has been fixed in 0.21 right? in that case should we not add the corresponding "fixed for 0.21" tag for this issue so we close it when that's released and just remove this from the 0.21 project board?

yorikvanhavre, 2023-05-08:
I understand it's the Qt version that's the problem, Qt5.15.7 crashes, Qt5.15.8 works.
On our side everything is okay then.. @kkremitzki it might be good to issue a 0.20.3 in Debian (Qt 5.15.8 is in unstable and testing already) when @donovaly is ready with his?

adrianinsaval, 2023-05-08:
could this also be related to the multi monitor bug on windows where freecad fails to launch? do we have an issue for that bug?

benj5378, 2024-09-01:
I'll just add these details
```
#95 0x00007ffff2cf7c1c in  () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#96 0x00007ffff2cfb234 in PyRun_StringFlags () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#97 0x00007ffff3790240 in Base::InterpreterSingleton::runString[abi:cxx11](char const*)
    (this=0x55555562c630, sCmd=0x7ffff6eaaba0 <FreeCADGuiInit> "#", '*' <repeats 75 times>, "\n#*   Copyright (c) 2002,2003 Jürgen Riegel <juergen.riegel@web.de>         *\n#*", ' ' <repeats 43 times>...)
    at /home/bensay/Development/FreeCAD/src/Base/Interpreter.cpp:249
#98 0x00007ffff645b590 in Gui::Application::runInitGuiScript() ()
    at /home/bensay/Development/FreeCAD/src/Gui/Application.cpp:1981
```

closed 2024-10-27 by adrianinsaval