FreeCAD/FreeCAD-Bundle #171: Mac OS conda weekly-builds not updated

rimkii, 2023-04-20:
I have been using the weekly conda development builds of the Mac version for some time.  However, over about the last month it seems like every time there is a new download, it has the same build number 32532 with a new date on it.  I acquire the download from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds.  Today every other build on that page is at rev 32929.  Is there a reason the Mac version is not at 32929.  I download the 32532 build with the 2023-0420 date, installed it but the about box still shows it as rev 32532.

[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.32532 (Git)
Build type: Release
Branch: master
Hash: 20e44eba50e1365b9e7bb9c5c244afe96a627a28
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.83.0
  * dodo
  * sheetmetal.backup1663254342.459754
  * sheetmetal.backup1663252767.516832
  * sheetmetal.backup1663253572.586379
  * sheetmetal.backup1663252689.521863
  * A2plus 0.4.59
  * fasteners 0.4.6
  * sheetmetal 0.2.57
  * sheetmetal.backup1663254477.573518
  * sheetmetal.backup1663270894.490148
[/code]

jbaehr, 2023-04-25:
@looooo could the reason for this be the python version? I've seen that the freecad_bundle job for mac passes without an error but the latest freecad that gets downloaded is always the same. I'm not completely sure where it is actually build, but the azure pipeline for the freecad-feedstock at conda-forge seems to build against python 3.11 while the bundle requests a freecad build against python 3.10. Maybe build of FreeCAD-0.21.0.32532 was just the latest one still targeting py310?

rimkii, 2023-04-30:
I have seen 2 new versions, so it looks like we are working again.  Thanks.

closed 2023-04-30 by rimkii