FreeCAD/FreeCAD-Bundle #173: [Issue] Cannot make meshes anymore

leoheck, 2023-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git) AppImage
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.0
  * Curves 0.6.8
  * fasteners 0.4.56
  * lattice2 1.0.0
  * freecad.gears 1.0.0
  * sheetmetal 0.2.63
  * Assembly4 0.50.2
  * kicadStepUpMod 10.18.8
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Issue description

Select Mesh WB
Select the Part to be converted to a mesh
Click Create Mesh from Shape
And then it throws this error.

The Mesh options can be edited, but only Standard and Gmsh tabs are clickable.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-08:
please copy and paste error messages instead of making screenshots

leoheck, 2023-05-09:
```
00:02:59  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/tmp/.mount_FreeCAslWo2w/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ImportError'>: /tmp/.mount_FreeCAslWo2w/usr/lib/././libnglib4smesh.so: undefined symbol: _ZNK17XCAFDoc_ColorTool8GetColorERK9TDF_LabelR14Quantity_Color

00:03:06  Traceback (most recent call last):
  File "<string>", line 5, in <module>
<class 'NameError'>: name 'MeshPart' is not defined

00:03:06  name 'MeshPart' is not defined
```

leoheck, 2023-05-21:
Where are the Releases of 0.21 Weekly AppImages? I cannot find them anymore.
I have this problem with Meshes that I want to see if it was fixed but I could not find those releases.

leoheck, 2023-05-22:
The latest version still has the same issue with the generation of MeshParts

```
02:52:01  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/tmp/.mount_FreeCAXxkiHA/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ImportError'>: /tmp/.mount_FreeCAXxkiHA/usr/lib/././libnglib4smesh.so: undefined symbol: _ZNK17XCAFDoc_ColorTool8GetColorERK9TDF_LabelR14Quantity_Color
02:52:03  Traceback (most recent call last):
  File "<string>", line 5, in <module>
<class 'NameError'>: name 'MeshPart' is not defined
02:52:03  name 'MeshPart' is not definedTraceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/tmp/.mount_FreeCAXxkiHA/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ImportError'>: /tmp/.mount_FreeCAXxkiHA/usr/lib/././libnglib4smesh.so: undefined symbol: _ZNK17XCAFDoc_ColorTool8GetColorERK9TDF_LabelR14Quantity_Color
02:52:09  Traceback (most recent call last):
  File "<string>", line 5, in <module>
<class 'NameError'>: name 'MeshPart' is not defined
02:52:09  name 'MeshPart' is not defined
```

```
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33263 (Git) AppImage
Build type: Release
Branch: master
Hash: 864f99bf984c237ef38d9451fd6f3a5ec34e6eb4
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.0
  * Curves 0.6.8
  * fasteners 0.4.56
  * lattice2 1.0.0
  * freecad.gears 1.0.0
  * sheetmetal 0.2.63
  * Assembly4 0.50.2
  * kicadStepUpMod 10.18.8
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
```

realthunder, 2023-05-22:
I have submitted patches ([here](https://github.com/conda-forge/smesh-feedstock/pull/69) and [here](https://github.com/looooo/SMESH/pull/1)) for this issue. But it seems @looooo is still busy. A temporary work around is to replace the `smesh` dependency ([here](https://github.com/conda-forge/freecad-feedstock/blob/b0af988408006e5b69e311d9af8f7cd1b50ae22a/recipe/meta.yaml#L60) and [here](https://github.com/conda-forge/freecad-feedstock/blob/b0af988408006e5b69e311d9af8f7cd1b50ae22a/recipe/meta.yaml#L81)) with my personal build of `smesh-rt`, if we have the write access to `freecad-feedstock`.

leoheck, 2023-05-22:
Oh, thank you for the feedback @realthunder, I am going to check your work there.

My temporary fix (for non-power users) is to have the stable 0.20 version installed too, so if I have to export meshes I just launch it instead.

adrianinsaval, 2023-05-22:
in the forum it was reported that FEM workbench also triggers this error:
https://forum.freecad.org/viewtopic.php?p=683490#p683490

kaktusus, 2023-05-25:
Next:
https://forum.freecad.org/viewtopic.php?p=683174#p683174

adrianinsaval, 2023-05-25:
In around 15 minutes there should be a new appimage bundle on the release page, please test it

adrianinsaval, 2023-05-26:
Do we know if the macOS version is affected too?

kaktusus, 2023-05-26:
fixed
```
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33303 (Git) AppImage
Build type: Release
Branch: master
Hash: 5b075a9938b78076fdde9034de4a05685e42690e
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
 ```

adrianinsaval, 2023-05-26:
Like I said above, the root of the problem is not really fixed yet, I've just delayed this

leoheck, 2023-05-26:
Yes, and as you said, we should test it, which was what @kaktusus did, and I also came here to say it worked. Thank you @adrianinsaval 

```
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33303 (Git) AppImage
Build type: Release
Branch: master
Hash: 5b075a9938b78076fdde9034de4a05685e42690e
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
```

adrianinsaval, 2023-05-26:
reported on macOS too now: [#9669](https://github.com/FreeCAD/FreeCAD/issues/9669) unfortunately looo's repo is failing to build on macOS now so I can't apply the same workaround

luzpaz, 2023-06-07:
bumping

adrianinsaval, 2023-06-07:
I'll request merge access to the relevant conda repos for us and merge realthunder's PR if they approve me

Adrián Insaurralde

adrianinsaval, 2023-06-18:
can somebody please test the latest macos release? @alexandervincon

I merged realthunder's PR to the smesh feedstock so hopefully this should be fixed now.

JohnOCFII, 2023-06-18:
> can somebody please test the latest macos release? @alexandervincon
> 
> I merged realthunder's PR to the smesh feedstock so hopefully this should be fixed now.

I'm not the OP, but tested as requested. Both the Mesh and FEM workbenches have issues with the latest Weekly Build for macOS.  Neither issue exists with my self-built version built with @oursland's Build Script (https://github.com/oursland/FreeCAD-Build-Notes)

Here's the info regarding the FreeCAD Weekly that exhibited the issues:

```
OS: macOS 13.4
Word size of FreeCAD: 64-bit
Version: 0.21.0.33411 (Git)
Build type: Release
Branch: master
Hash: 127404d793d5c2081383b46a7b677426d9a0c5aa
Python 3.11.4, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
```

And the issues:

Mesh:
```
16:54:29  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Applications/Other Apps/FreeCAD-Intel-33411.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ImportError'>: dlopen(/Applications/Other Apps/FreeCAD-Intel-33411.app/Contents/Resources/lib/MeshPart.so, 0x0002): Symbol not found: __ZNK17XCAFDoc_ColorTool8GetColorERK9TDF_LabelR14Quantity_Color
  Referenced from: <388A345A-BA59-33E8-99F7-BD253D5BB832> /Applications/Other Apps/FreeCAD-Intel-33411.app/Contents/Resources/lib/libnglib4smesh.dylib
  Expected in:     <24B29773-5A02-3D45-883F-DCD16A0F3606> /Applications/Other Apps/FreeCAD-Intel-33411.app/Contents/Resources/lib/libTKXCAF.7.7.1.dylib
```

FEM:
[image]

JohnOCFII, 2023-06-19:
I also rebuilt my local copy to ensure that the current build didn't introduce any issues.  It still worked fine, so the problem is something to do with the build/dependancies.

```
    - OS: macOS 13.4
    - Word size of FreeCAD: 64-bit
    - Version: 0.21.0.33411 (Git)
    - Build type: Release
    - Branch: master
    - Hash: 127404d793d5c2081383b46a7b677426d9a0c5aa
    - Python 3.11.0, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.6.3
    - Locale: English/United States (en_US)
```

alexandervincon, 2023-06-19:
> can somebody please test the latest macos release? @alexandervincon
> 
> I merged realthunder's PR to the smesh feedstock so hopefully this should be fixed now.

issue still existing

adrianinsaval, 2023-06-26:
I'm working on getting conda builds using occt 7.6.3 working again, once that is done I'll change the bundles to use those builds and this should get fixed. We'll still have to deal with this later on conda-forge though, eventually we will want to move to occt 7.7

leoheck, 2023-06-26:
Is it hard, do you want some help on that?

vernzimm, 2023-06-26:
Thanks all for working on this. I'll be waiting for new weekly build when it becomes available again 😄

adrianinsaval, 2023-06-26:
> Is it hard, do you want some help on that?

Thank you! On getting it to build with 7.6 I got it working already, just some more tweaking left to do before putting it live, but for the long run there is a couple remaining issues that I was not able to solve but am working around for the time being. For some reason I can't get this to cross compile on macos, it keeps complaining that the compiler is for arm64-apple-darwin but it wants it to be aarch64-apple-darwin (which should mean the same in this situation), and on windows it just won't compile with boost 1.78:
https://dev.azure.com/adrianinsaval/feedstock-builds/_build/results?buildId=20&view=logs&j=4b3757f9-99c5-5891-6c0c-93d7030cdf7d&t=f047df50-a233-5ae9-bf8f-c7dd8e5684ad
my workarounds is to use a separate CI that has mac m1 available to compile that natively (this is likely actually better but probably not doable for the official conda-forge repo, this is low priority). And for the windows problem since I downgraded to occt already downgrading to boost 1.74 was also required and with that it works, but in the future we want this to work.

If you find out how to fix those that would be helpful. f you are not familiar with the system there's quite some reading and experimentation to do to get the hang of it, or at least it took me a while, keywords are conda-forge, mamba, conda-build, conda-smithy. Some repos to look into: https://github.com/conda-forge/freecad-feedstock https://github.com/FreeCAD/freecad-feedstock and the mess I'm doing at https://github.com/adrianinsaval/freecad-feedstock

adrianinsaval, 2023-07-13:
please test latest macos bundles

JohnOCFII, 2023-07-13:
> please test latest macos bundles

Today's arm bundle seems to have resolved the Mesh and FEM issues. (At least I can create a Mesh, and I can launch the FEM module and choose some actions -- I don't really know FEM well enough to test beyond that, and it is further than I could get with the testing previously as noted above in this thread.)

```
    - OS: macOS 13.4.1
    - Word size of FreeCAD: 64-bit
    - Version: 0.21.0.33576 (Git)
    - Build type: Release
    - Branch: master
    - Hash: 8e6fbd2886d3ad51b75232bd0019d92d496e80cb
    - Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
    - Locale: C/Default (C)
```

closed 2023-07-25 by adrianinsaval