FreeCAD/FreeCAD-Bundle #176: GPU is not being used in FreeCAD weekly conda Windows releases.

ftdt10, 2023-05-23:
Freecad weekly builds conda Windows version doesn't use GPU in Windows.
In Task Manager, I confirmed that GPU is not being used, only the CPU is being used.
So, FPS is very slow in FreeCAD weekly release. For reference, it's normal on Ubuntu.  

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33263 (Git)
Build type: Release
Branch: master
Hash: 864f99bf984c237ef38d9451fd6f3a5ec34e6eb4
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * Curves 0.6.9
  * kicadStepUpMod 10.19.1
[/code]

ftdt10, 2023-05-23:
[image]

adrianinsaval, 2023-05-23:
please don't double post issues... this is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/9642
closing for the same reason...

closed 2023-05-23 by adrianinsaval

adrianinsaval, 2023-05-23:
see #164 instead for the error message you mentioned