FreeCAD/FreeCAD-Bundle #179: Add 'Conda' to FreeCAD About info dialog of macOS

luzpaz, 2023-05-30:
IMHO we avoid an extra layer of abstraction by adding 'Conda' to macOS About info. See usecase in https://github.com/FreeCAD/FreeCAD/issues/9692

Currently on the dev version:
```
OS: macOS 11.7.7
Word size of FreeCAD: 64-bit
Version: 0.21.0.33321 (Git)
Build type: Release
Branch: master
```

Proposal:  
```
OS: macOS 11.7.7 (Conda)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33321 (Git) 
Build type: Release
Branch: master
```

JohnOCFII, 2023-05-30:
Is `Conda` vs. `Other` the real issue here, or is this specific to the automated weekly builds (via Conda) vs. potentially self-built versions, which could be Conda, Homebrew, etc?

luzpaz, 2023-05-30:
Good question, what are your thoughts on this ?

luzpaz, 2023-05-30:
@adrianinsaval do you mind weighing-in on this ?

adrianinsaval, 2023-05-30:
Yes this could be useful, there's already a cmake variable related to conda I think so we could use that to add this at compile time

luzpaz, 2023-05-30:
How would cmake 'know' where from/whom is compiling it (in scenarios where people are compiling FreeCAD from source)?

adrianinsaval, 2023-05-30:
Sorry I hadn't read all the comments, I think that we are hitting a point of diminishing returns on how much info we gather and output, if someone can compile themselves they probably can tell us that they did and using what. As for identifying conda builds, we set this cmake variable for our builds and most people building on conda would be doing this too:
https://github.com/FreeCAD/FreeCAD/blob/ef70f2e53de65a49aa8e4077732353d4b4dcec02/cMake/FreeCAD_Helpers/InitializeFreeCADBuildOptions.cmake#L13
https://github.com/conda-forge/freecad-feedstock/blob/b0af988408006e5b69e311d9af8f7cd1b50ae22a/recipe/build.sh#L48

oursland, 2023-05-30:
Having a Conda identifier is somewhat useful.  Conda is different from other build systems in that it packages a custom set of frameworks and standard libraries that are likely incompatible with the ones on the host system.  The reason is to provide a consistent set of base APIs independent of the macOS version.

I learned the hard way that if you somehow mix up linkage between a Conda library and a host library provided by macOS or some other package manager like `brew`, your binary will run but crash in unexpected ways.  Having some indicator to identify if this is a possibility may be valuable for someone trying to read a bug report or crash logs.

closed 2024-11-11 by adrianinsaval

luzpaz, 2024-11-11:
Thank you @adrianinsaval :+1: