FreeCAD/FreeCAD-Bundle #180: Crash on macOS when filesystem is case sensitive (icudtl.dat and qtwebengine* nor found)

AKielhorn, 2023-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 11.7.7
Word size of FreeCAD: 64-bit
Version: 0.21.0.33321 (Git)
Build type: Release
Branch: master
Hash: 1d61ed187c9633702818c6cc5d7a1f07dd052ca8
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * Assembly4 0.12.6
  * Plot 2022.4.17
  * BOLTSFC 2022.11.5
  * MnesarcoUtils 0.2.1
  * A2plus 0.4.60k
  * fcgear
  * FreeCAD_assembly3 0.11.3
  * fasteners 0.4.55
  * lattice2 1.0.0
  * sheetmetal 0.2.61
[/code]
```

### Subproject(s) affected?

None

### Problem description

FreeCAD on macOS crashes as described here:

[MacOS at startup crash (critical)](https://forum.freecad.org/viewtopic.php?t=71835)

The error message is:

```
Qt WebEngine resources not found at /Applications/FreeCAD.app/Contents/Resources/Resources. Trying parent directory...
```

When I look into the application I see

```
/Applications/FreeCAD.app/Contents/Resources/resources
```

with a lowercase "r".

If I rename the folder to "Resources" it works.

This affects only a small number of macOS users, since the default on macOS is to ignore case.

Greetings
Axel

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-05-31:
Moved to FreeCAD-Bundle where conda related issues are relevant.

adrianinsaval, 2023-06-06:
I don't quite understand why you where missing that link, the bundling process is supposed to add it

AKielhorn, 2023-06-09:
When I look into the distributed application on the disc image I see:

```
% cd /Volumes/FreeCAD_weekly-builds-33321-2023-05-28-conda-macOS-x86_64-py311/FreeCAD.app/Contents/Resources 
Resources % ls -l 
total 1392
drwxr-xr-x     5 axel  staff     160 28 Mai 14:25 Ext/
-rw-rw-r--     1 axel  staff    1082 16 Nov  2019 LICENSE
drwxr-xr-x    30 axel  staff     960 28 Mai 14:25 Mod/
drwxr-xr-x     9 axel  staff     288 28 Mai 14:30 bin/
drwxr-xr-x    10 axel  staff     320 28 Mai 14:25 cmake/
drwxr-xr-x   245 axel  staff    7840 28 Mai 14:30 conda-meta/
drwxr-xr-x     4 axel  staff     128 28 Mai 14:24 doc/
drwxr-xr-x     6 axel  staff     192 28 Mai 14:24 etc/
drwxr-xr-x    33 axel  staff    1056 28 Mai 14:23 fonts/
-rw-r--r--     1 axel  staff  280788 28 Mai 14:17 freecad-doc.icns
-rw-r--r--     1 axel  staff   98455 28 Mai 14:17 freecad-script.icns
-rw-r--r--     1 axel  staff  319441 28 Mai 14:17 freecad.icns
drwxr-xr-x  2026 axel  staff   64832 28 Mai 14:30 lib/
drwxr-xr-x     6 axel  staff     192 28 Mai 14:30 libexec/
drwxr-xr-x     4 axel  staff     128 28 Mai 14:24 man/
drwxr-xr-x    79 axel  staff    2528 28 Mai 14:24 mkspecs/
drwxr-xr-x    15 axel  staff     480 28 Mai 14:24 phrasebooks/
drwxr-xr-x    31 axel  staff     992 28 Mai 14:24 plugins/
drwxr-xr-x    28 axel  staff     896 28 Mai 14:25 qml/
-rw-r--r--     1 axel  staff      30 28 Mai 14:17 qt.conf
drwxr-xr-x     8 axel  staff     256 28 Mai 14:30 resources/
drwxr-xr-x    14 axel  staff     448 28 Mai 14:24 sbin/
drwxr-xr-x    65 axel  staff    2080 28 Mai 14:25 share/
drwxr-xr-x     9 axel  staff     288 28 Mai 14:24 ssl/
drwxr-xr-x   348 axel  staff   11136 28 Mai 14:25 translations/
drwxr-xr-x     3 axel  staff      96 28 Mai 14:24 var/
```

There is no link to `Resources`.

Maybe the distribution is build on a case insensitive file system, or the disc image isn't case sensitive.
When I start FreeCAD from a case insensitive file system, it works.

On a case insensitive file system, the link fails:
```
Resources % ln -s resources Resources 
ln: Resources/resources: File exists
```

There is no simple solution that works in both cases.

luzpaz, 2023-06-10:
What's the solution then ?

adrianinsaval, 2023-06-10:
I would guess we have to build the package in a case sensitive file system, but I don't know if that is available on github actions. Or find out why it searching for the wrong directory to begin with. I've never used mac though so I don't know even where to begin looking into this

luzpaz, 2023-06-10:
@ipatch have you encountered this problem with homebrew ?

ipatch, 2023-06-10:
@luzpaz i do remember running across an issue where something was case sensitive and it was not working due to things being case sensitive. that said, i've slept several times since that was reported, but i do remember commenting on a github issue and possibly a forum thread about it.

unfortunately i am going to be AFK for the 10 days so i won't be able to test much if at all, but when i get back i don't mind taking a look. there is still a lot of things i'd to resolve with the homebrew-freecad tap as well.

AKielhorn, 2023-06-11:
Even if you build on a case sensitive file system, there will be a conflict when users install the application on a case insensitive file system, which is the default for macOS.

For me everything works fine when I have only the `Resources` folder.
Renaming instead of linking may be a solution.

adrianinsaval, 2023-07-09:
Can you check if this is still a problem in the latest weekly?

AKielhorn, 2023-07-09:
I checked the latest weekly:

[code]
OS: macOS 11.7.8
Word size of FreeCAD: 64-bit
Version: 0.21.0.33492 (Git)
Build type: Release
Branch: master
Hash: 6e235a4809e14674aa2cd23e68630b68998a53d4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Assembly4 0.12.6
  * Plot 2022.4.17
  * BOLTSFC 2022.11.5
  * MnesarcoUtils 0.2.1
  * A2plus 0.4.60k
  * fcgear
  * FreeCAD_assembly3 0.11.3
  * fasteners 0.4.55
  * lattice2 1.0.0
  * sheetmetal 0.2.61
[/code]

and it works fine.

Thaks for the fix.

Axel

closed 2023-07-10 by adrianinsaval