FreeCAD/FreeCAD-Bundle #181: ARM build 31483 is not working under macOS Sonoma

neatNuseful, 2023-06-11:
Hi,

The following build is unfortunately not working under macOS Sonoma. I'm using a MacBook Pro M2 Max. On Ventura 13.4, it is working fine.

During installation, a message appear that the application requires rosetta 2. Then other error message that the application cannot open.
FreeCAD_weekly-builds-31483-2022-12-31-conda-macOS-arm-py311.dmg

luzpaz, 2023-06-11:
Full [About](https://wiki.freecad.org/About) info please

mecProject, 2023-06-11:
It is not possible since the app didn't launch.
[image]

mecProject, 2023-06-11:
I also tried to open it from terminal (also clicked open FreeCAD in privacy and security settings). This is what I get:

The application /Applications/FreeCAD.app cannot be opened for an unexpected reason, error=Error Domain=RBSRequestErrorDomain Code=5 "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed., NSUnderlyingError=0x600001dee190 {Error Domain=NSPOSIXErrorDomain Code=162 "Unknown error: 162" UserInfo={NSLocalizedDescription=Launchd job spawn failed}}}

luzpaz, 2023-06-12:
@PLChris can you reproduce on macOS ?

makialnooh, 2023-06-12:
@mecProject I am getting the very same message .. any solution yet?

mecProject, 2023-06-12:
No, it is still not working. I hope this can be solved soon.

JohnOCFII, 2023-06-13:
One Comment

macOS Sonoma is an early Beta of a release that isn't likely to ship until fall 2023.

https://en.wikipedia.org/wiki/MacOS_Sonoma#:~:text=The%20successor%20to%20macOS%20Ventura,located%20in%20California%27s%20Sonoma%20County.

adrianinsaval, 2023-06-26:
does this new build work? https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-33447-2023-06-26-conda-macOS-arm-py310.dmg

mecProject, 2023-06-26:
unfortunately not. same issue.

adrianinsaval, 2023-06-26:
Ok, is apple providing any documentation on what is changing on that release? Are older apps even expected to be compatible at this moment?

JohnOCFII, 2023-06-26:
> Ok, is apple providing any documentation on what is changing on that release? Are older apps even expected to be compatible at this moment?

An Apple-focused developer builds their work life around the annual WWDC (World Wide Developers Conference) which just happened a few weeks back.  There Apple has a couple of general sessions covering the highpoints of changes (Platform State of the Union is key here).  After that, they have many, many breakout videos on new features.   All videos are free.  During this week early developer Betas are released, with the final new OS versions available in late Fall - October timeframe.  During this 5-month period developers usually download the new Beta OS releases and start testing.  Most apps aren't expected to actually work against the new Beta's until late in the cycle.  It is not uncommon for large applications to not work with the new OS releases for a few months AFTER final release in the fall.  https://developer.apple.com/wwdc23/

mecProject, 2023-06-26:
Actually, I didn't have any issue with other apps like Blender, Inkscape, GIMP. They are working fine.

Velaba, 2023-06-28:
> Hi,
>
> The following build is unfortunately not working under macOS Sonoma. I'm using a MacBook Pro M2 Max. On Ventura 13.4, it is working fine.
>
> During installation, a message appear that the [...]

I was having this issue and couldn't figure out why. I hadn't been on the app in a little while so I did a fresh download, but ran the same issue. 

It didn't dawn on me that it could be a macOS Sonoma beta problem at first, so I right clicked the application in finder, selected "Show Packaged Contents" and within the macOS directory, I found a unix executable file. I ran that and waited a little bit and sure enough, I got the freeCAD window open. 

I hope this helps in case someone hasn't already found another solution to get it working.

PLowran, 2023-06-28:
Yo! @Velaba 
This Worked for Me 100%!!

adrianinsaval, 2023-06-29:
could it be because the app is not signed?

mecProject, 2023-06-29:
@Velaba solution works fine for me. Thanks!

cbenhagen, 2023-07-09:
This is happens as well with the x86_64 version. Code signing and notarization do not seem to be the issue. I signed the app bundle with my certificate and get the following output:
```codesign --verify --deep --verbose /Applications/FreeCAD.app
/Applications/FreeCAD.app: valid on disk
/Applications/FreeCAD.app: satisfies its Designated Requirement
```

```
spctl --assess --verbose /Applications/FreeCAD.app
/Applications/FreeCAD.app: accepted
source=Developer ID
```

Here are some relevant log files though:
```
AMFI: Launch Constraint Violation (enforcing), error info: c[1]p[1]m[1]e[13], (Constraint not matched) launching proc[vc: 1 pid: 19150]: /usr/bin/env, launch type 3, failure proc [vc: 1 pid: 19150]: /usr/bin/env
ASP: Security policy would not allow process: 19150, /Applications/FreeCAD.app/Contents/MacOS/FreeCAD
```

@adrianinsaval any ideas? Do you have a macOS 14 machine to test?

adrianinsaval, 2023-07-09:
Unfortunately I don't have a mac

edgardmello, 2023-07-21:
You guys can use Raycast app with the resource quick link using this path /Applications/FreeCAD.app/Contents/MacOS/FreeCAD and open with terminal

This will do the job and open freecad by terminal with no issues and faster than going to finder, right click, etc etc.

lagnat, 2023-07-25:
This is resolved with the latest Sonoma beta.

adrianinsaval, 2023-07-26:
Excellent, I'm closing the issue then.

closed 2023-07-26 by adrianinsaval