FreeCAD/FreeCAD-Bundle #186: Version: 0.21.0.33458 crashing when saving a file

tonealone, 2023-06-30:
Version info:
```
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
```
Attached are crash logs:

[freecad-2023-06-29-210548.txt](https://github.com/FreeCAD/FreeCAD-Bundle/files/11911508/freecad-2023-06-29-210548.txt)
[freecad-2023-06-29-210331.txt](https://github.com/FreeCAD/FreeCAD-Bundle/files/11911509/freecad-2023-06-29-210331.txt)
[freecad-2023-06-29-204533.txt](https://github.com/FreeCAD/FreeCAD-Bundle/files/11911510/freecad-2023-06-29-204533.txt)
[freecad-2023-06-29-202712.txt](https://github.com/FreeCAD/FreeCAD-Bundle/files/11911511/freecad-2023-06-29-202712.txt)
[freecad-2023-06-29-191438.txt](https://github.com/FreeCAD/FreeCAD-Bundle/files/11911512/freecad-2023-06-29-191438.txt)

zisoft, 2023-07-06:
Same problem here. I tried with a complete fresh installation by renaming the folders `~/Library/Application Support/FreeCad` and `~/Library/Preferences/FreeCad`

```
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33492 (Git)
Build type: Release
Branch: master
Hash: 6e235a4809e14674aa2cd23e68630b68998a53d4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
```

Starting FreeCad from terminal gives the following output, segfault on save:

```
FreeCAD 0.21.0, Libs: 0.21.0R33492 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Populating font family aliases took 182 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
[11029:18947:0706/184725.728396:ERROR:command_buffer_proxy_impl.cc(141)] ContextResult::kTransientFailure: Failed to send GpuChannelMsg_CreateCommandBuffer.
2023-07-06 18:47:30.793 freecad[11026:220301] +[CATransaction synchronize] called within transaction
2023-07-06 18:47:31.163 freecad[11026:220301] +[CATransaction synchronize] called within transaction
[1]    11026 segmentation fault  ./freecad --log-file /Users/mario/freecad.log
```

zisoft, 2023-07-06:
There is already a bug tracker: https://github.com/FreeCAD/FreeCAD/issues/9860

luzpaz, 2023-07-19:
@tonealone this is a packaging repository. The main FreeCAD bugtracker is at https://github.com/FreeCAD/FreeCAD/issues

closed 2023-07-19 by luzpaz