FreeCAD/FreeCAD-Bundle #187: Incorrect generation of indexes in windows array (parametric count)

kaiwas, 2023-07-04:
Hi!
An array of windows that counts the number of array elements according to some rules (wall length) does not correctly assign indexes to array elements.
The error occurs both when the wall length increases and when the wall length decreases. But a certain logic could not be detected.

Temporary solution to the problem.  To assign in the array settings Expand Array - > false.
Save the file. Close it and open it.
Then do Expand Array - > True.

Video example 
https://youtu.be/PbrzLWttXDM

File https://disk.yandex.ru/d/vKtR90BAvnkA1w

paullee0, 2023-07-10:
Discussion on Forum for reference
-  https://forum.freecad.org/viewtopic.php?t=79566

(original thread and post - https://forum.freecad.org/viewtopic.php?t=79555#p692320)

adrianinsaval, 2023-07-13:
how is this bundle related? This repo is exclusviely about bundling freecad packages, the source code of the actual application is at https://github.com/FreeCAD/FreeCAD and general problems with the applications should be dealt with there.