FreeCAD/FreeCAD-Bundle #195: Freecad Weekly does not have the AppImage is not working

leoheck, 2023-08-05:
It was working fine after my PR. Now it is not right anymore.

I removed everything from my system.
Then I downloaded just the Weekly build. This one:
```
FreeCAD_weekly-builds-33682-2023-08-03-conda-Linux-x86_64-py310_0f0268a3695a2e958624cfbc18cd36ca.AppImage
```

Then after installing it with the [AppImage Launcher](https://github.com/TheAssassin/AppImageLauncher) as I always do. 
I am seeing this. The wrong icon image for the icon launcher and a problematic name. 
[image]

If I have to install the stable version, both launchers will have the same icon and the same text.

I am also providing here an extra diagnosis, of this AppImage to show you extra information using `./appimagelint-x86_64.AppImage`:
```bash
~/Applications …
➜ ./appimagelint-x86_64.AppImage FreeCAD_weekly-builds-33682-2023-08-03-conda-Linux-x86_64-py310_0f0268a3695a2e958624cfbc18cd36ca.AppImage 
appimagelint.cli[675722] [INFO] Checking AppImage FreeCAD_weekly-builds-33682-2023-08-03-conda-Linux-x86_64-py310_0f0268a3695a2e958624cfbc18cd36ca.AppImage
appimagelint.cli[675722] [INFO] Running check "GNU libc ABI check"
appimagelint.glibc_abi_check[675722] [INFO] detected required version for runtime: 2.14
appimagelint.glibc_abi_check[675722] [INFO] detected required version for payload: 2.33
appimagelint.glibc_abi_check[675722] [INFO] [✖] AppImage can run on Debian oldstable (bullseye)
appimagelint.glibc_abi_check[675722] [INFO] [✔] AppImage can run on Debian stable (bookworm)
appimagelint.glibc_abi_check[675722] [INFO] [✔] AppImage can run on Debian testing (trixie)
appimagelint.glibc_abi_check[675722] [INFO] [✔] AppImage can run on Debian unstable (sid)
appimagelint.glibc_abi_check[675722] [INFO] [✔] AppImage can run on Ubuntu lunar
appimagelint.glibc_abi_check[675722] [INFO] [✔] AppImage can run on Ubuntu jammy
appimagelint.glibc_abi_check[675722] [INFO] [✖] AppImage can run on Ubuntu focal
appimagelint.glibc_abi_check[675722] [INFO] [✖] AppImage can run on Ubuntu bionic
appimagelint.glibc_abi_check[675722] [INFO] [✖] AppImage can run on Ubuntu xenial
appimagelint.glibc_abi_check[675722] [INFO] [✖] AppImage can run on Ubuntu trusty
appimagelint.glibc_abi_check[675722] [INFO] [✖] AppImage can run on CentOS 7
appimagelint.cli[675722] [INFO] Running check "GNU libstdc++ ABI check"
appimagelint.glibcxx_abi_check[675722] [INFO] detected required version for runtime: <none>
appimagelint.glibcxx_abi_check[675722] [INFO] detected required version for payload: 3.4.30
appimagelint.glibcxx_abi_check[675722] [INFO] [✖] AppImage can run on Debian oldstable (bullseye)
appimagelint.glibcxx_abi_check[675722] [INFO] [✖] AppImage can run on Debian stable (bookworm)
appimagelint.glibcxx_abi_check[675722] [INFO] [✖] AppImage can run on Debian testing (trixie)
appimagelint.glibcxx_abi_check[675722] [INFO] [✖] AppImage can run on Debian unstable (sid)
appimagelint.glibcxx_abi_check[675722] [INFO] [✔] AppImage can run on Ubuntu lunar
appimagelint.glibcxx_abi_check[675722] [INFO] [✔] AppImage can run on Ubuntu jammy
appimagelint.glibcxx_abi_check[675722] [INFO] [✖] AppImage can run on Ubuntu focal
appimagelint.glibcxx_abi_check[675722] [INFO] [✖] AppImage can run on Ubuntu bionic
appimagelint.glibcxx_abi_check[675722] [INFO] [✖] AppImage can run on Ubuntu xenial
appimagelint.glibcxx_abi_check[675722] [INFO] [✖] AppImage can run on Ubuntu trusty
appimagelint.glibcxx_abi_check[675722] [INFO] [✖] AppImage can run on CentOS 7
appimagelint.cli[675722] [INFO] Running check "Icons validity and location check"
appimagelint.icons_check[675722] [INFO] Extracting icon name from desktop file: /tmp/.mount_FreeCA28VJM1/freecad.desktop
appimagelint.icons_check[675722] [WARNING] Multiple matching icons found in AppDir root, checking all
appimagelint.icons_check[675722] [INFO] Checking resolution of icon: /tmp/.mount_FreeCA28VJM1/freecad.png
appimagelint.icons_check[675722] [INFO] Checking resolution of icon: /tmp/.mount_FreeCA28VJM1/freecad.svg
appimagelint.icons_check[675722] [INFO] [✔] Valid icon in AppDir root
appimagelint.icons_check[675722] [INFO] Checking resolution of icon: /tmp/.mount_FreeCA28VJM1/.DirIcon
appimagelint.icons_check[675722] [INFO] [✔] Valid icon file in .DirIcon
appimagelint.icons_check[675722] [INFO] Checking resolution of icon: /tmp/.mount_FreeCA28VJM1/usr/share/icons/hicolor/16x16/apps/freecad.png
appimagelint.icons_check[675722] [INFO] Checking resolution of icon: /tmp/.mount_FreeCA28VJM1/usr/share/icons/hicolor/32x32/apps/freecad.png
appimagelint.icons_check[675722] [INFO] Checking resolution of icon: /tmp/.mount_FreeCA28VJM1/usr/share/icons/hicolor/48x48/apps/freecad.png
appimagelint.icons_check[675722] [INFO] Checking resolution of icon: /tmp/.mount_FreeCA28VJM1/usr/share/icons/hicolor/64x64/apps/freecad.png
appimagelint.icons_check[675722] [WARNING] Icon found whose file name doesn't match the Icon= entry in desktop file: hicolor/64x64/apps/freecad_weekly.png
appimagelint.icons_check[675722] [INFO] Checking resolution of icon: /tmp/.mount_FreeCA28VJM1/usr/share/icons/hicolor/scalable/apps/freecad.svg
appimagelint.icons_check[675722] [WARNING] Icon found whose file name doesn't match the Icon= entry in desktop file: hicolor/scalable/apps/freecad_weekly.svg
appimagelint.icons_check[675722] [WARNING] Icon found whose file name doesn't match the Icon= entry in desktop file: hicolor/scalable/apps/org.freecadweb.FreeCAD.svg
appimagelint.icons_check[675722] [WARNING] Icon found whose file name doesn't match the Icon= entry in desktop file: hicolor/scalable/mimetypes/application-x-extension-fcstd.svg
appimagelint.icons_check[675722] [INFO] [✔] Other integration icons valid
appimagelint.cli[675722] [INFO] Running check "Desktop files existence and validity"
appimagelint.desktop_files[675722] [INFO] Checking desktop files in root directory
appimagelint.desktop_files[675722] [INFO] [✖] Exactly one desktop file in AppDir root
appimagelint.desktop_files[675722] [INFO] Checking desktop file /tmp/.mount_FreeCAMvYc4t/freecad.desktop with desktop-file-validate
/tmp/.mount_FreeCAMvYc4t/freecad.desktop: hint: value "Engineering;" for key "Categories" in group "Desktop Entry" does not contain a registered main category; application might only show up in a "catch-all" section of the application menu
appimagelint.desktop_files[675722] [INFO] Checking desktop file /tmp/.mount_FreeCAMvYc4t/freecad_weekly.desktop with desktop-file-validate
/tmp/.mount_FreeCAMvYc4t/freecad_weekly.desktop: hint: value "Engineering;" for key "Categories" in group "Desktop Entry" does not contain a registered main category; application might only show up in a "catch-all" section of the application menu
appimagelint.desktop_files[675722] [INFO] Checking desktop file /tmp/.mount_FreeCAMvYc4t/usr/share/applications/freecad_weekly.desktop with desktop-file-validate
/tmp/.mount_FreeCAMvYc4t/usr/share/applications/freecad_weekly.desktop: hint: value "Engineering;" for key "Categories" in group "Desktop Entry" does not contain a registered main category; application might only show up in a "catch-all" section of the application menu
appimagelint.desktop_files[675722] [INFO] Checking desktop file /tmp/.mount_FreeCAMvYc4t/usr/share/applications/org.freecadweb.FreeCAD.desktop with desktop-file-validate
/tmp/.mount_FreeCAMvYc4t/usr/share/applications/org.freecadweb.FreeCAD.desktop: warning: value "Система автоматизированного проектирования" for key "Comment[ru]" in group "Desktop Entry" looks the same as that of key "GenericName[ru]"
/tmp/.mount_FreeCAMvYc4t/usr/share/applications/org.freecadweb.FreeCAD.desktop: hint: value "Graphics;Science;Education;Engineering;" for key "Categories" in group "Desktop Entry" contains more than one main category; application might appear more than once in the application menu
/tmp/.mount_FreeCAMvYc4t/usr/share/applications/org.freecadweb.FreeCAD.desktop: hint: value "Graphics;Science;Education;Engineering;" for key "Categories" in group "Desktop Entry" contains more than one main category; application might appear more than once in the application menu
/tmp/.mount_FreeCAMvYc4t/usr/share/applications/org.freecadweb.FreeCAD.desktop: warning: value "application/x-extension-fcstd;model/obj;model/iges;image/vnd.dwg;image/vnd.dxf;model/vnd.collada+xml;application/iges;model/iges;model/step;model/step+zip;model/stl;application/vnd.shp;model/vrml;" for key "MimeType" in group "Desktop Entry" contains "model/iges" more than once
appimagelint.desktop_files[675722] [INFO] [✔] All desktop files in AppDir are valid
```

And this is the output with colors:
[image]

I would like to ask to at least have a launcher that has a different icon and text for the weekly build and the stable version.

chennes, 2023-08-05:
@adrianinsaval

adrianinsaval, 2023-08-05:
This is because your second PR and now it gets confused because there's two desktop files on the appimage when it's supposed to have only one. I will push my commit with a fix for the weeklies tomorrow. Not sure if we should republish the stable appimage for this

adrianinsaval, 2023-08-16:
can you try the latest appimage @leoheck ?

leoheck, 2023-08-17:
Sure, let me check.

leoheck, 2023-08-17:
This new `FreeCAD_weekly-builds-33798-2023-08-16-conda-Linux-x86_64-py310.AppImage`, shows this image right after the installation. I don't have to do any dumb as the normal Windows-like approach such as restarting the computer or logout. It looks correct, just out of the box. Thank you!
[image]

Here is the output of the command line tool that replaces the old app image, which had a bad launcher/icon
``` 
> Updating FreeCAD_weekly-builds-33798-2023-08-13-conda-Linux-x86_64-py310.AppImage
Starting update...
Validation warning: AppImage not signed
Removing old AppImage: /home/lheck/Applications/FreeCAD_weekly-builds-33798-2023-08-13-conda-Linux-x86_64-py310.AppImage
Update successful. New file created: /home/lheck/Applications/FreeCAD_weekly-builds-33798-2023-08-16-conda-Linux-x86_64-py310.AppImage
Checking for updates...
Fetching release information for tag "weekly-builds" from GitHub API.
Fetching release information for tag "weekly-builds" from GitHub API.
zsync2: Using CA bundle found on system: /etc/ssl/certs/ca-certificates.crt
... done!

Fetching release information for tag "weekly-builds" from GitHub API.
0% done
Updating from GitHub Releases via ZSync
Fetching release information for tag "weekly-builds" from GitHub API.
zsync2: Using CA bundle found on system: /etc/ssl/certs/ca-certificates.crt
0% done
zsync2: Target file: /home/lheck/Applications/FreeCAD_weekly-builds-33798-2023-08-16-conda-Linux-x86_64-py310.AppImage
zsync2: Reading seed file: FreeCAD_weekly-builds-33798-2023-08-13-conda-Linux-x86_64-py310.AppImage
99.63% done (953.23 of 956.78 MiB)...
zsync2: Usable data from seed files: 99.838325%
zsync2: Renaming temp file
zsync2: Fetching remaining blocks
99.84% done (955.23 of 956.78 MiB)...
zsync2: Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/93114989/5a8748ff-1dd5-4e6e-beff-28399c7fc78c?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230817%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230817T000950Z&X-Amz-Expires=300&X-Amz-Signature=1824a38a9b224142772a7e27b093830dfea2664d95162d9a213c8b0ce09a4d7f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=93114989&response-content-disposition=attachment%3B%20filename%3DFreeCAD_weekly-builds-33798-2023-08-16-conda-Linux-x86_64-py310.AppImage&response-content-type=application%2Foctet-stream
zsync2: optimized ranges, old requests count 8, new requests count 6

99.87% done (955.52 of 956.78 MiB)...
zsync2: Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/93114989/5a8748ff-1dd5-4e6e-beff-28399c7fc78c?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230817%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230817T000950Z&X-Amz-Expires=300&X-Amz-Signature=1824a38a9b224142772a7e27b093830dfea2664d95162d9a213c8b0ce09a4d7f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=93114989&response-content-disposition=attachment%3B%20filename%3DFreeCAD_weekly-builds-33798-2023-08-16-conda-Linux-x86_64-py310.AppImage&response-content-type=application%2Foctet-stream
zsync2: optimized ranges, old requests count 5, new requests count 4

99.87% done (955.52 of 956.78 MiB)...
zsync2: Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/93114989/5a8748ff-1dd5-4e6e-beff-28399c7fc78c?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230817%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230817T000950Z&X-Amz-Expires=300&X-Amz-Signature=1824a38a9b224142772a7e27b093830dfea2664d95162d9a213c8b0ce09a4d7f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=93114989&response-content-disposition=attachment%3B%20filename%3DFreeCAD_weekly-builds-33798-2023-08-16-conda-Linux-x86_64-py310.AppImage&response-content-type=application%2Foctet-stream
zsync2: optimized ranges, old requests count 5, new requests count 4

99.87% done (955.52 of 956.78 MiB)...
zsync2: Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/93114989/5a8748ff-1dd5-4e6e-beff-28399c7fc78c?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230817%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230817T001158Z&X-Amz-Expires=300&X-Amz-Signature=cbfc142095b673bf4435932dbd786dbf48ac451179168f22c0ad45ffeab23372&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=93114989&response-content-disposition=attachment%3B%20filename%3DFreeCAD_weekly-builds-33798-2023-08-16-conda-Linux-x86_64-py310.AppImage&response-content-type=application%2Foctet-stream
zsync2: optimized ranges, old requests count 5, new requests count 4

100.00% done (956.74 of 956.78 MiB)...
zsync2: Verifying downloaded file
100.00% done (956.78 of 956.78 MiB)...
zsync2: checksum matches OK
zsync2: used 1001635840 local, fetched 1750208
100.00% done (956.78 of 956.78 MiB)...
```

Now a small question regarding the image. Why did you change the image that was in use before? This one does not have nice art of the conda symbol. The previous one was created by me too and it was never a problem. 

Now, I want to have the stable version too... let me do a quick check on what happens when I install it too.
Alright, here it is. I think we are back where we were a couple of days ago. But at least the weekly has its own image and label. Hope it remains like this after the reboot... haha
[image]

@adrianinsaval thank you. Do you plan to push this to the stable version as well?
The stable version could have `0.21` on its name like `Freecad 0.21` too.

adrianinsaval, 2023-08-17:
Versioning on the desktop file is not planned, @chennes do you think this is worth reissuing the stable appimage? I can easily push the change to fix it there too.

> Now a small question regarding the image. Why did you change the image that was in use before? This one does not have nice art of the conda symbol. The previous one was created by me too and it was never a problem.

the svg looked weird on my pc:
[image]

so I didn't want to risk it and got an svg of the anaconda icon which seems to be a full circle not a C, I'm totally open to changing that icon if you don't think it looks good.

leoheck, 2023-08-17:
It was just a question. If you think it is good and it was not working for you then this new one is fine for me. I don't know much about conda, but a quick Google showed me these 2 logos. I used the first one, since a C, is the first letter of Conda and I feel is better. But that is my personal taste. But I also feel that a bit of white is missing in the current version.
[image]

adrianinsaval, 2023-08-17:
ok so the icon I used is from anaconda.org and the C seems to be the icon for the conda package manager itself, a little confusing but oh well. I re did the previous icon to try to get it to work properly and found some weird empty elements that once removed made the icon work:
[image]

closed 2023-08-17 by adrianinsaval