FreeCAD/FreeCAD-Bundle #200: Error in. open freecad.exe

amorfiniluigi82, 2023-11-22:
Hi, my name luigi from italy .. download freecad version: FreeCAD_weekly-builds-35081-2023-11-20-conda-Windows-x86_64-py310.7z
with freecad 0.21.1 installed.. my show error:

[image]

My freecad installed:
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * A2plus 0.4.61
  * Curves 0.6.14
  * Fasteners 0.4.70
  * gears 1.0.0
  * Pyramids-and-Polyhedrons
  * SheetMetal 0.3.12
[/code]

Idea?
Thanks.

YosypC, 2023-11-28:
The message translates as "The procedure entry point [...] could not be located in the dynamic link library [...]"

luzpaz, 2023-11-28:
@adrianinsaval any idea ?

adrianinsaval, 2023-11-28:
I suspect this is solved. Possibly related to coin 4.0.1 changes. @amorfiniluigi82 can you retry on the latest bundle?

luzpaz, 2023-11-29:
@amorfiniluigi82 please report back

closed 2024-01-19 by adrianinsaval