FreeCAD/FreeCAD-Bundle #201: The preferences window doesn't fit properly in 1366 x 768 displays

P-dos-Santos, 2023-11-25:
Hello folks. As described above, we have an issue with the preferences windows that cannot apply any new settings because the button to confirm them is hidden from the display. 

See the following images.

[image]
[image]

Thank you

P-dos-Santos, 2023-11-25:
```
OS: Fedora Linux 38 (KDE Plasma) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35176 (Git) AppImage
Build type: Release
Branch: main
Hash: dcd533af4bf0cb3dbc50c7d0e2fb579cad168819
Python 3.10.13, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
```

JohnOCFII, 2023-11-25:
Appears to be a duplicate of this (which has already been acknowledged as a known issue): https://github.com/FreeCAD/FreeCAD/issues/11004#issuecomment-1825880450

amorfiniluigi82, 2023-11-25:
Idem from from Linux AppImage.

[image]

[image]

chennes, 2023-11-25:
This is not a Bundle issue. Closing in favor of FreeCAD#11511

closed 2023-11-25 by chennes