FreeCAD/FreeCAD-Bundle #204: [Bug] On Sketcher Workbench, the tools Polyline and B-spline are not disabled when pressing esc key.

rainell, 2023-12-27:
[Bug] On MacOS 10.15.7 (Catalina), FreeCAD 0.22.0dev, Sketcher Workbench, the tools Polyline and B-spline are not disabled when pressing esc key.

JohnOCFII, 2023-12-28:
I'm not seeing that issue on macOS Sonoma.  Are you running any tools that modify the window handling, such as Magnet or Rectangle?  If so, that might be related.   Those apps usually have an option to white-list Apps so that they won't impact the app in question.  If that is something you are using, you might try to whitelist FreeCAD.  
```
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35456 (Git)
Build type: Release
Branch: main
Hash: c91a35961f7ef42f1b64eb11ca963d6fbc0b42da
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

rainell, 2023-12-29:
@JohnOCFII the problem is verified in macOS 10.15.7 (Catalina), with the latest 0.22.0dev builds and I don't know exactly from which build it started happening, but it did not happen in previous builds. I'm not running any tools that modify the window handling.

JohnOCFII, 2023-12-30:
> @JohnOCFII the problem is verified in macOS 10.15.7 (Catalina), with the latest 0.22.0dev builds and I don't know exactly from which build it started happening, but it did not happen in previous builds. I'm not running any tools that modify the window handling.

Yes, I see the issue now.  I had seen something similar a few weeks back when exiting the new constraints option.  A workaround is to select one of the other sketch items (such as rectangle). and then press `ESC.`

Hopefully one of the FreeCAD developers will see this.  I know the last similar item I reported was a quick fix.

```
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35456 (Git)
Build type: Release
Branch: main
Hash: c91a35961f7ef42f1b64eb11ca963d6fbc0b42da
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

adrianinsaval, 2024-01-19:
fixed: https://github.com/FreeCAD/FreeCAD/pull/12018
please report these issues in the mian repo not here. this is only for packaging problems

closed 2024-01-19 by adrianinsaval