FreeCAD/FreeCAD-Bundle #208: Assembly4>Animator>Save complains OpenCV not installed

ppemawm, 2024-01-18:
@adrianinsaval
Assembly4 animator save function complains that CV2 is not installed and prevents saving the animation when in fact CV2 is installed.
Error occurs in certain stable and weekly releases referenced in discussions below.

Please see discussions here:
https://forum.freecad.org/viewtopic.php?p=733824#p733824
https://forum.freecad.org/viewtopic.php?p=732390#p732390
https://forum.freecad.org/viewtopic.php?p=726914#p726914

adrianinsaval, 2024-01-18:
As mentioned in the forum thread opencv doesn't seem to be the problem if it is incorrectly reported by asm4. The actual problem is with pillow that has removed qt5 support in version 10.0

adrianinsaval, 2024-01-19:
uploaded pillow 9.5 pacakges to conda: https://anaconda.org/freecad/pillow/files
new bundles should be up shortly and I confirmed at least the appimage works:
[image]

closed 2024-01-19 by adrianinsaval