FreeCAD/FreeCAD-Bundle #210: Material workbench content is missing in recent Windows weekly builds

FEA-eng, 2024-01-25:
I've noticed that recent weekly builds for Windows are missing the content of the Material workbench and the following error occurs:

```
19:56:23  DLL load failed while importing MatGui: The specified module could not be found.
19:56:23  Traceback (most recent call last):
  File "E:\FreeCAD_weekly-builds-35720-2024-01-23-conda-Windows-x86_64-py310\Mod\Material\InitGui.py", line 41, in Initialize
    import MatGui
  File "E:\FreeCAD_weekly-builds-35720-2024-01-23-conda-Windows-x86_64-py310\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
```

This also makes the FEM workbench unusable since materials can't be added.

I talked about that with @davesrocketshop and he suggested that build logs should be checked for missing DLL. I don't know who is responsible for Windows builds so I'm just reporting this here.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35720 (Git)
Build type: Release
Branch: main
Hash: 00cd50bbf49c810555899efe112ec7595c76cd3a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

davesrocketshop, 2024-01-25:
To clarify, the Material.DLL is missing content, which impacts the MatGui.DLL.

You should be able to see  the following:
`
>>> import Material
>>> dir(Material)
['Material', 'MaterialManager', 'Model', 'ModelManager', 'ModelProperty', 'UUIDs', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__']
>>>
 `

instead, this is seen:
`['__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__']`

maxwxyz, 2024-03-19:
Is this still relevant?

FEA-eng, 2024-03-19:
> Is this still relevant?

Currently, it's ok so I'll close this issue.

closed 2024-03-19 by FEA-eng