FreeCAD/FreeCAD-Bundle #211: Cannot create a projection

napwodz, 2024-03-04:
Hi.

When trying to create a drawing of a step model, I cannot create a set of projections - I get an error:

> 10:29:10  pyException: Traceback (most recent call last):
>  File "<string>", line 1, in <module>
> <class 'Base.FreeCADError'>: vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 3)

Whatever file I try to work on, I get the same error. A ProjGroup is created, but when I double-click on it to edit, I get Segfault and the following message:

> Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7fcd0645b050]
#1  0x7fccc59be08b in TechDrawGui::TaskProjGroup::saveGroupState() from /tmp/.mount_FreeCAOYWKB5/usr/lib/TechDrawGui.so+0x1bb
#2  0x7fccc59bee1d in TechDrawGui::TaskProjGroup::TaskProjGroup(TechDraw::DrawProjGroup*, bool) from /tmp/.mount_FreeCAOYWKB5/usr/lib/TechDrawGui.so+0xb3d
#3  0x7fccc59bf3c2 in TechDrawGui::TaskDlgProjGroup::TaskDlgProjGroup(TechDraw::DrawProjGroup*, bool) from /tmp/.mount_FreeCAOYWKB5/usr/lib/TechDrawGui.so+0x62
#4  0x7fccc5b0e748 in TechDrawGui::ViewProviderProjGroup::setEdit(int) from /tmp/.mount_FreeCAOYWKB5/usr/lib/TechDrawGui.so+0x78
#5  0x7fccc5b0e4bf in TechDrawGui::ViewProviderProjGroup::doubleClicked() from /tmp/.mount_FreeCAOYWKB5/usr/lib/TechDrawGui.so+0xf
#6  0x7fcd08b8171d in Gui::TreeWidget::mouseDoubleClickEvent(QMouseEvent*) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libFreeCADGui.so+0x39d
#7  0x7fcd06faf111 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Widgets.so.5+0x6b1
#8  0x7fcd0705656e in QFrame::event(QEvent*) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Widgets.so.5+0x1e
#9  0x7fcd060b3d5b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Core.so.5+0x7b
#10  0x7fcd06f7034e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#11  0x7fcd06f78222 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#12  0x7fcd08925f89 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libFreeCADGui.so+0x99
#13  0x7fcd060b3fc8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Core.so.5+0x108
#14  0x7fcd06f763ab in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#15  /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e39) [0x7fcd06fc6e39]
#16  /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Widgets.so.5(+0x1c9f99) [0x7fcd06fc9f99]
#17  0x7fcd06f7035e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#18  0x7fcd08925f89 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libFreeCADGui.so+0x99
#19  0x7fcd060b3fc8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Core.so.5+0x108
#20  0x7fcd06734b7c in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Gui.so.5+0x7dc
#21  0x7fcd0670ad7c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Gui.so.5+0xac
#22  /tmp/.mount_FreeCAOYWKB5/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x6454a) [0x7fcd00c7254a]
#23  /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/./libglib-2.0.so.0(+0x5a417) [0x7fcd04911417]
#24  /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/./libglib-2.0.so.0(+0x5d3c7) [0x7fcd049143c7]
#25  /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7fcd049149be]
#26  0x7fcd061083d6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Core.so.5+0x66
#27  0x7fcd060b2bbb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Core.so.5+0x11b
#28  0x7fcd060ba6a2 in QCoreApplication::exec() from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libQt5Core.so.5+0x82
#29  0x7fcd088899c7 in Gui::Application::runApplication() from /tmp/.mount_FreeCAOYWKB5/usr/bin/../lib/libFreeCADGui.so+0xa27
#30  /tmp/.mount_FreeCAOYWKB5/usr/bin/freecad(+0x7620) [0x559107b88620]
#31  /lib/x86_64-linux-gnu/libc.so.6(+0x2724a) [0x7fcd0644624a]
#32  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7fcd06446305]
#33  /tmp/.mount_FreeCAOYWKB5/usr/bin/freecad(+0x7981) [0x559107b88981]

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36228 (Git) AppImage
Build type: Release
Branch: main
Hash: 6c8ee5964521a99e01c8d33cbf9aac8a19273ef3
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * btl 0.9.9
  * FeedsAndSpeeds 0.5.0
  * InventorLoader 1.4.0
  * sheetmetal 0.3.13
```
```

napwodz, 2024-03-04:
Duplicate - please, remove

maxwxyz, 2024-03-19:
@napwodz is this still an issue with the current weekly build?

napwodz, 2024-04-22:
Sorry for late reply. 

The problem lied in negative value of the "LineStandard" parameter and AFAIK it has been solved with: https://github.com/FreeCAD/FreeCAD/commit/6337b454e1

Thanks!

closed 2024-04-22 by maxwxyz