FreeCAD/FreeCAD-Bundle #213: PolarPattern cannot be applied to Additive Helix

Superbobo75, 2024-03-06:
In version 0.22.0Dev 36228 I am trying to create this model according to this video. The problem occurs when I try to perform PolarPattern on Additive Helix which has the values below. All based on a tutorial video on youtube where it has no problem.

[Video Tutorial](https://www.youtube.com/watch?v=Q0VqEOnQkTs)

[image]

Attempting 8 occurences in PolarPattern. Up to 3 occurences it's ok, on the fourth it starts to fall apart. 

[image]

I can create the same model in 21.1 just fine. When I save the model and open it in the Dev version, the model displays correctly and PolarPattern can be edited without errors. I can increase and decrease occurrences. 
But if I edit Helix, or re-create it after deleting Polar and Helix, Polar starts behaving with the above error.

maxwxyz, 2024-03-19:
@Superbobo75 is this still an issue in the current weekly builds https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds 
If not I'll close this, if this still exist, I'll move this issue to the FreeCAD repo.

Superbobo75, 2024-03-20:
@maxwxyz  It already works in the current version.

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Czech/Czech Republic (cs_CZ)
Installed mods: 
  * Assembly4 0.50.7
  * BIM 2021.12.0
  * Curves 0.6.23
  * Manipulator 1.5.7
  * ThreadProfile 1.89.0