FreeCAD/FreeCAD-Bundle #214: Cannot open my files anymore

leoheck, 2024-03-09:
Hey guys, I installed this version `FreeCAD_weekly-builds-36331-2024-03-08-conda-Linux-x86_64-py310.AppImage`, yesterday but now I can't open my files. Someone else with the same issue? Where can I get the previews weekly to test with it again?

Can you have the current weekly and one previous on the releases page so we can get back to the old one when the last one does not work anymore?

zu4lu, 2024-03-12:
Yes, same issue here

leoheck, 2024-03-12:
I fixed it by reusing an older version of my files. I used to work on a
Dropbox folder, so I didn't lose much. But I had to redo my last changes.

This kind of issue is a bit of worry if you are doing important things.

leoheck, 2024-03-18:
@zu4lu could you share your history here? Could you open your file(s)?
https://github.com/FreeCAD/FreeCAD/issues/13034

adrianinsaval, 2024-03-19:
What do you mean when you say you can't open your files? Your issue us to vague

leoheck, 2024-03-19:
Freecad segfauls when I open this files saved with the weekly builds built last week or a couple of days ago. I have 4 files that Freecad cannot open anymore since it crashes immediately when I do File > Open those files. The last 4 releases of the weekly builds are kind of doing this. They are quite unstable. I The weekly builds were never as unstable as those are. When I am on my PC I can get some logs while trying to open those files.

maxwxyz, 2024-03-19:
closed in favor of https://github.com/FreeCAD/FreeCAD/issues/13034

closed 2024-03-19 by maxwxyz