FreeCAD/FreeCAD-Bundle #216: Development Version  FreeCAD_weekly-builds-36391-2024-03-11-conda-Windows-x86_64-py310.7z Problems When Rows Deleted in Spreadsheet

BrianAtDocumentedDesigns, 2024-03-12:
I am new to FreeCad (maybe 3 months). Many of my drawings are pretty simple and driven off spreadsheets. The last two development versions I downloaded if you delete a row from the spreadsheet the model develops all sorts of errors. I suspect the "link" between alias and spread sheet cells gets broken (get "null face in AttachEngine3D ...)

xtemp09, 2024-03-19:
Why don't you copy-and-paste those errors and provide steps to reproduce them?

----

I've just created alias and deleted a row. I got:

> 14:03:32  <Spreadsheet> Cell.cpp(638): Unnamed#Spreadsheet.B1: Property 'my_alias1' not found in 'my_alias1'
> 14:03:32  Recompute failed!

Do you mean these messages?

BrianAtDocumentedDesigns, 2024-03-19:
Sorry I tried to copy the messages but WxWidgets (I assume) doesn't let you copy certain things and got the message I typed consistently. I had forgot about the console window (?) as I have it closed all the time. If I get the errors again I will do so but it sounds like you have the same issue.

xtemp09, 2024-03-19:
These messages are normal. 😅  I'd recommend to close this issue.

BrianAtDocumentedDesigns, 2024-03-19:
I don't see how it is normal if it doesn't happen with 0.21

xtemp09, 2024-03-19:
If you look at the source code of _Spreadsheet_, no change has been made for a long time. The commits are: removed whitespace, unnecessary functions, modernization of C++ and so on.

> the console window

Do you mean _Report view_? Could you turn it on and reproduce the error?

maxwxyz, 2024-03-19:
Closing this issue. If there is still an issue in the current builds, please open a problem report at the FreeCAD repo: https://github.com/FreeCAD/FreeCAD/issues