FreeCAD/FreeCAD-Bundle #217: Can't run in Arch Linux - Appimage - FreeCAD_weekly-builds-36480-2024-03-18-conda-Linux-x86_64-py311.AppImage

erickwill, 2024-03-18:
Here is the log after running in Arch Linux (rolling release):

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3c770) [0x758b3985a770]
#1  0x758b3cc40913 in Gui::MDIView::buildWindowTitle() from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADGui.so+0x23
#2  0x758b3cbc9eec in Gui::MainWindow::_updateActions() from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADGui.so+0x6c
#3  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libQt5Core.so.5(+0x2e9214) [0x758b39ce9214]
#4  0x758b39cecc8a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libQt5Core.so.5+0x2a
#5  0x758b39cddfad in QObject::event(QEvent*) from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libQt5Core.so.5+0x7d
#6  0x758b3ab7035e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#7  0x758b3c727399 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADGui.so+0x99
#8  0x758b39cb3fc8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libQt5Core.so.5+0x108
#9  0x758b39d078f1 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libQt5Core.so.5+0x381
#10  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libQt5Core.so.5(+0x3080c4) [0x758b39d080c4]
#11  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x758b38310ad7]
#12  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x758b38313ac7]
#13  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x758b383140ce]
#14  0x758b39d083d6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libQt5Core.so.5+0x66
#15  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADGui.so(+0xa02922) [0x758b3cc02922]
#16  0x758b3b2f8555 in Base::ConsoleSingleton::notifyPrivate(Base::LogStyle, Base::IntendedRecipient, Base::ContentType, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADBase.so+0x75
#17  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADBase.so(+0x108a39) [0x758b3b308a39]
#18  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADBase.so(+0xf433a) [0x758b3b2f433a]
#19  0x758b3b2f498a in Base::ConsoleSingleton::sPyLog(_object*, _object*) from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADBase.so+0x4a
#20  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libpython3.11.so.1.0(+0x205a58) [0x758b3be05a58]
#21  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x25c) [0x758b3bde559c]
#22  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x758b3bdf0f52]
#23  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libpython3.11.so.1.0(+0x2aa9ad) [0x758b3beaa9ad]
#24  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x758b3beaa424]
#25  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libpython3.11.so.1.0(+0x2c9893) [0x758b3bec9893]
#26  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libpython3.11.so.1.0(+0x2c532a) [0x758b3bec532a]
#27  /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x758b3beb9669]
#28  0x758b3b33ba1c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADBase.so+0x6c
#29  0x758b3c66a193 in Gui::Application::runInitGuiScript() from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADGui.so+0x43
#30  0x758b3c7554f1 in Gui::StartupPostProcess::showMainWindow() from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADGui.so+0xe1
#31  0x758b3c7572a4 in Gui::StartupPostProcess::execute() from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADGui.so+0x64
#32  0x758b3c68a756 in Gui::Application::runApplication() from /tmp/.mount_FreeCAEVoWQw/usr/bin/../lib/libFreeCADGui.so+0x4d6
#33  /tmp/.mount_FreeCAEVoWQw/usr/bin/freecad(+0x7620) [0x59d0665e1620]
#34  /usr/lib/libc.so.6(+0x25cd0) [0x758b39843cd0]
#35  /usr/lib/libc.so.6(__libc_start_main+0x8a) [0x758b39843d8a]
#36  /tmp/.mount_FreeCAEVoWQw/usr/bin/freecad(+0x7981) [0x59d0665e1981]

maxwxyz, 2024-03-19:
Should be fixed with https://github.com/FreeCAD/FreeCAD/pull/13033

@adrianinsaval could you rerun the jobs as this is the third ticket now in a few hours.

adrianinsaval, 2024-03-19:
> Should be fixed with https://github.com/FreeCAD/FreeCAD/pull/13033
> 
> @adrianinsaval could you rerun the jobs as this is the third ticket now in a few hours. 

Yes I'll make a new build shortly

closed 2024-03-19 by maxwxyz