FreeCAD/FreeCAD-Bundle #218: Windows 11 Threat blocked: FreeCAD_weekly-builds-36596-2024-03-25-conda-Windows-x86_64-py311.7z

erickwill, 2024-03-26:
From a couple of builds ago, Windows Security start to flag the FreeCad zip as a threat. Initially, I thought it could be a false positive, but after so many builds it's still detecting as a threat. 
Please verify this issue.

`Threat blocked

Detected: Trojan:Script/Wacatac.B!ml
Status: Removed
A threat or app was removed from this device.

Details: This program is dangerous and executes commands from an attacker.

Affected items: file: C:\Users\Erick\Apps\FreeCAD_weekly-builds-36596-2024-03-25-conda-Windows-x86_64-py311.7z

webfile: C:\Users\Erick\AppsFreeCAD_weekly-builds-36596-2024-03-25-conda-Windows-x86_64-py311.7z|https://objects.githubusercontent.com/github-production-release-asset-2e65be/93114989/6311c99d-9d38-4c9a-acc6-8548a4c02750?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240318%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240318T234709Z&X-Amz-Expires=300&X-Amz-Signature=f07037abb7d505c754717dd9a1e033a345e7d71c33ce542f65122175c2689bc7&X-Amz-SignedHeaders=host&actor_id=549381&key_id=0&repo_id=93114989&response-content-disposition=attachment%3B%20filename%3DFreeCAD_weekly-builds-36596-2024-03-25-conda-Windows-x86_64-py311.7z&response-content-type=application%2Foctet-stream|pid:4088,ProcessStart:133552792541316408
`

sclebo05, 2024-03-26:
Had this issue as well, but it cleared up the past few builds

sclebo05, 2024-03-26:
Downloaded and extracted the latest (FreeCAD_weekly-builds-36596-2024-03-25-conda-Windows-x86_64-py311.7z) with no issue.  Make sure your Windows Defender updates are the latest?

adrianinsaval, 2024-03-31:
I've heard that windows defender and other antiviruses are now flagging anything that uses python that isn't signed as malware, but I don't know. I haven't experienced these issues myself

adrianinsaval, 2024-07-05:
not much we can do about MS bs and I have not seen more reports about this

closed 2024-07-05 by adrianinsaval