FreeCAD/FreeCAD-Bundle #220: [Linux] Weekly-builds issue 2024.04.23: xkbcommon: ERROR: failed to add default include path /home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/X11/xkb

erickwill, 2024-04-23:
Found this error trying to run the weekly-build of 04/23/2024 release, appimage.

` Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

xkbcommon: ERROR: failed to add default include path /home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/X11/xkb
(qt.qpa.wayland) failed to create xkb context
QOpenGLFunctions created with non-current context
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3c770) [0x77ae8ee5a770]
#1  /tmp/.mount_FreeCAXSOjWM/usr/plugins/platforms/../../lib/libxkbcommon.so.0(xkb_context_ref+0) [0x77ae89628280]
#2  /tmp/.mount_FreeCAXSOjWM/usr/plugins/platforms/../../lib/libxkbcommon.so.0(+0x1e775) [0x77ae8962a775]
#3  /tmp/.mount_FreeCAXSOjWM/usr/plugins/platforms/../../lib/libxkbcommon.so.0(xkb_keymap_new_from_buffer+0x47) [0x77ae89629f37]
#4  0x77ae8974c508 in QtWaylandClient::QWaylandInputDevice::Keyboard::keyboard_keymap(unsigned int, int, unsigned int) from /tmp/.mount_FreeCAXSOjWM/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0xe4
#5  /tmp/.mount_FreeCAXSOjWM/usr/plugins/platforms/../../lib/./libffi.so.8(+0x6a4a) [0x77ae8a145a4a]
#6  /tmp/.mount_FreeCAXSOjWM/usr/plugins/platforms/../../lib/./libffi.so.8(+0x5fea) [0x77ae8a144fea]
#7  /tmp/.mount_FreeCAXSOjWM/usr/plugins/platforms/../../lib/libwayland-client.so.0(+0xa0f8) [0x77ae899f80f8]
#8  /tmp/.mount_FreeCAXSOjWM/usr/plugins/platforms/../../lib/libwayland-client.so.0(+0x6949) [0x77ae899f4949]
#9  /tmp/.mount_FreeCAXSOjWM/usr/plugins/platforms/../../lib/libwayland-client.so.0(wl_display_dispatch_queue_pending+0x74) [0x77ae899f6124]
#10  0x77ae8975099b in QtWaylandClient::QWaylandDisplay::flushRequests() from /tmp/.mount_FreeCAXSOjWM/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0x2d
#11  0x77ae897565ab in QtWaylandClient::QWaylandWindow::setVisible(bool) from /tmp/.mount_FreeCAXSOjWM/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0x127
#12  0x77ae8f13eeb7 in QWindowPrivate::setVisible(bool) from /tmp/.mount_FreeCAXSOjWM/usr/bin/../lib/libQt5Gui.so.5+0x77
#13  0x77ae8f9a3ba1 in QWidgetPrivate::show_sys() from /tmp/.mount_FreeCAXSOjWM/usr/bin/../lib/libQt5Widgets.so.5+0x221
#14  0x77ae8f9ac10b in QWidgetPrivate::show_helper() from /tmp/.mount_FreeCAXSOjWM/usr/bin/../lib/libQt5Widgets.so.5+0x6b
#15  0x77ae8f9ae7c3 in QWidgetPrivate::setVisible(bool) from /tmp/.mount_FreeCAXSOjWM/usr/bin/../lib/libQt5Widgets.so.5+0x183
#16  0x77ae91bcb68c in Gui::MainWindow::startSplasher() from /tmp/.mount_FreeCAXSOjWM/usr/bin/../lib/libFreeCADGui.so+0x25c
#17  0x77ae91755252 in Gui::StartupPostProcess::showMainWindow() from /tmp/.mount_FreeCAXSOjWM/usr/bin/../lib/libFreeCADGui.so+0x142
#18  0x77ae91756fe4 in Gui::StartupPostProcess::execute() from /tmp/.mount_FreeCAXSOjWM/usr/bin/../lib/libFreeCADGui.so+0x64
#19  0x77ae9168a036 in Gui::Application::runApplication() from /tmp/.mount_FreeCAXSOjWM/usr/bin/../lib/libFreeCADGui.so+0x4d6
#20  /tmp/.mount_FreeCAXSOjWM/usr/bin/freecad(+0x7620) [0x60c97c8c7620]
#21  /usr/lib/libc.so.6(+0x25cd0) [0x77ae8ee43cd0]
#22  /usr/lib/libc.so.6(__libc_start_main+0x8a) [0x77ae8ee43d8a]
#23  /tmp/.mount_FreeCAXSOjWM/usr/bin/freecad(+0x7981) [0x60c97c8c7981]
`

Let me know if any other information is necessary.

rmu75, 2024-04-23:
I got the same (modulo addresses / tmp file names etc) traceback with current weekly build. This is on KDE neon / ubuntu 22.04 LTS on a  wayland session.

backtrace from gdb:

```
Thread 1 "freecad" received signal SIGSEGV, Segmentation fault.
xkb_context_ref (ctx=0x0) at ../src/context.c:192
192     ../src/context.c: Datei oder Verzeichnis nicht gefunden.
(gdb) bt
#0  xkb_context_ref (ctx=0x0) at ../src/context.c:192
#1  0x00007fffeebe3775 in xkb_keymap_new (ctx=<optimized out>, format=<optimized out>, flags=flags@entry=XKB_KEYMAP_COMPILE_NO_FLAGS) at ../src/keymap-priv.c:67
#2  0x00007fffeebe2f37 in xkb_keymap_new_from_buffer (ctx=<optimized out>, 
    buffer=0x7fffb8f1d000 "xkb_keymap {\nxkb_keycodes \"(unnamed)\" {\n\tminimum = 8;\n\tmaximum = 708;\n\t<ESC>", ' ' <wiederholt 16 Mal>, "= 9;\n\t<AE01>", ' ' <wiederholt 15 Mal>, "= 10;\n\t<AE02>", ' ' <wiederholt 15 Mal>, "= 11;\n\t<AE03>", ' ' <wiederholt 15 Mal>, "= 12;\n\t<AE04>", ' ' <wiederholt 12 Mal>..., length=74181, format=<optimized out>, flags=<optimized out>) at ../src/keymap.c:193
#3  0x00007fffeed17508 in QtWaylandClient::QWaylandInputDevice::Keyboard::keyboard_keymap(unsigned int, int, unsigned int) () from /home/robert/fctmp/squashfs-root/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5
#4  0x00007fffef6e3a4a in ffi_call_unix64 () from /home/robert/fctmp/squashfs-root/usr/plugins/platforms/../../lib/./libffi.so.8
#5  0x00007fffef6e2fea in ffi_call_int () from /home/robert/fctmp/squashfs-root/usr/plugins/platforms/../../lib/./libffi.so.8
#6  0x00007fffeec160f8 in wl_closure_invoke (closure=0x555555737e00, flags=<optimized out>, target=<optimized out>, opcode=0, data=<optimized out>) at ../src/connection.c:1025
#7  0x00007fffeec12949 in dispatch_event (display=display@entry=0x5555556e2640, queue=0x5555556e2730) at ../src/wayland-client.c:1631
#8  0x00007fffeec14124 in dispatch_queue (queue=0x5555556e2730, display=0x5555556e2640) at ../src/wayland-client.c:1777
#9  wl_display_dispatch_queue_pending (display=0x5555556e2640, queue=0x5555556e2730) at ../src/wayland-client.c:2019
#10 0x00007fffeed1b99b in QtWaylandClient::QWaylandDisplay::flushRequests() () from /home/robert/fctmp/squashfs-root/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5
#11 0x00007fffeed215ab in QtWaylandClient::QWaylandWindow::setVisible(bool) () from /home/robert/fctmp/squashfs-root/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5
#12 0x00007ffff473eeb7 in QWindowPrivate::setVisible(bool) () from /home/robert/fctmp/squashfs-root/usr/bin/../lib/libQt5Gui.so.5
#13 0x00007ffff4fa3ba1 in QWidgetPrivate::show_sys() () from /home/robert/fctmp/squashfs-root/usr/bin/../lib/libQt5Widgets.so.5
#14 0x00007ffff4fac10b in QWidgetPrivate::show_helper() () from /home/robert/fctmp/squashfs-root/usr/bin/../lib/libQt5Widgets.so.5
#15 0x00007ffff4fae7c3 in QWidgetPrivate::setVisible(bool) () from /home/robert/fctmp/squashfs-root/usr/bin/../lib/libQt5Widgets.so.5
#16 0x00007ffff71cb68c in Gui::MainWindow::startSplasher() () from /home/robert/fctmp/squashfs-root/usr/bin/../lib/libFreeCADGui.so
#17 0x00007ffff6d55252 in Gui::StartupPostProcess::showMainWindow() () from /home/robert/fctmp/squashfs-root/usr/bin/../lib/libFreeCADGui.so
#18 0x00007ffff6d56fe4 in Gui::StartupPostProcess::execute() () from /home/robert/fctmp/squashfs-root/usr/bin/../lib/libFreeCADGui.so
#19 0x00007ffff6c8a036 in Gui::Application::runApplication() () from /home/robert/fctmp/squashfs-root/usr/bin/../lib/libFreeCADGui.so
#20 0x000055555555b620 in main ()

```

erickwill, 2024-04-23:
yep, well remembered. I'm also using kde neon, wayland.

luzpaz, 2024-04-23:
@adrianinsaval have you seen this before ?

adrianinsaval, 2024-04-23:
I experienced this on the Wayland session now too. I'm tried adding Wayland support to the appimage but looks like just adding qt-wayland to the bundle isn't enough. I'll revert that and continue testing

artelse, 2024-04-23:
Same here, is there an archive of older weekly builds?

adrianinsaval, 2024-05-03:
is this still a problem?

rmu75, 2024-05-03:
For me, current weekly build FreeCAD_weekly-builds-37083-conda-Linux-x86_64-py311.appimage does start and seems to mostly work, but I suspect is uses xcb backend  / Xwayland and it throws spurious 
`<class 'AttributeError'>: 'PySide2.QtWidgets.QWidget' object has no attribute 'unpolish'` errors, but that may be a problem in the Path / CAM module.

adrianinsaval, 2024-05-03:
> but I suspect is uses xcb backend / Xwayland

yes it is, I'll explore building a wayland appimage again later this month but for now we'll remain with x11

> it throws spurious
> `<class 'AttributeError'>: 'PySide2.QtWidgets.QWidget' object has no attribute 'unpolish'` errors, but that may be a problem in the Path / CAM module.

yeah I don't think that has anything to do with wayland, you should try starting freecad with logging enabled to see where the problem is coming from and report that to the main repot

closed 2024-05-03 by adrianinsaval