FreeCAD/FreeCAD-Bundle #223: Add latest rev_number available as weekly bundle in a file.

GAZ082, 2024-05-01:
Hello. Working in an update script to check and download the latest weekly bundle and found that the freecad_version.txt picks up the latest commit, which not necessarily is  the commit of the latest weekly bundle.

Would be handy to have a file named like "freecad_latest_weekly_version.txt" with the latest revision available as bundle.

Thanks!

adrianinsaval, 2024-05-02:
do you need the commit hash? the bundle name includes the revision number, you could just query that

GAZ082, 2024-05-02:
> do you need the commit hash? the bundle name includes the revision number, you could just query that

Hey Adrián,

I'm using the rev number, but the thing is, the freecad_version.txt file gets updated hours before the new build.

See for example now, we are already in 37086:
```
rev_number: 37086 (Git)
branch_name: main
commit_date: 2024/05/02 06:09:39
commit_hash: 368b91e0358dafbf17bf0c133dbff00343fd3422
remote_url: https://github.com/FreeCAD/FreeCAD main
```
and the latest build has the rev 37083, for example: FreeCAD_weekly-builds-**37083**-conda-Linux-x86_64-py311.AppImage]

So either sync the freecad_version.txt with the build time or have a new file with the latest revision that **has a build**.

adrianinsaval, 2024-05-02:
you'll notice there is a source package along side the binary builds, freecad_version.txt has information about the source package not the binary builds. The source package is updated very frequently while the binaries are updated normally twice a week.

My question is, why do you need to parse a txt? why not just use the revision number form the file name?

GAZ082, 2024-05-02:
> are you aware of github's API? for example you could parse https://api.github.com/repos/FreeCAD/FreeCAD-Bundle/releases/tags/weekly-builds to get the available revision number

Ahh no I didnt, this is awesome. Thanks!

closed 2024-05-02 by GAZ082