FreeCAD/FreeCAD-Bundle #227: FreeCAD breaks Python

ei14, 2024-05-11:
I use FreeCAD. I also program in Python. In my case, these are separate things, never mixed. Yet when I run `help("modules")` in a Python environment, Python crashes. I determined this is because `import freecad` causes a crash.

I do not want to use FreeCAD functionality in any of my Python programs. But I also do not want to uninstall the entirety of FreeCAD just in order to use the offline Python documentation reader `pydoc`. FreeCAD should not break my coding environment! It's a CAD software! What gives?

Frankly, why does FreeCAD even pollute python3.XX/site-packages to begin with? It's a GUI software that, from a user perspective, should have nothing to do with Python programming as far as I know.

Here is the Python output of `import freecad`:

```
Python 3.12.3 (main, Apr 23 2024, 09:16:07) [GCC 13.2.1 20240417] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import freecad
PATH_TO_FREECAD_LIBDIR not specified, using default FreeCAD version in /usr/lib/freecad/lib
Fatal Python error: PyImport_AppendInittab: PyImport_AppendInittab() may not be called after Py_Initialize()
Python runtime state: initialized

Current thread 0x00007934b9e72b80 (most recent call first):
  File "<frozen importlib._bootstrap>", line 488 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1289 in create_module
  File "<frozen importlib._bootstrap>", line 813 in module_from_spec
  File "<frozen importlib._bootstrap>", line 921 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1331 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1360 in _find_and_load
  File "/usr/lib/python3.12/site-packages/freecad/__init__.py", line 20 in <module>
  File "<frozen importlib._bootstrap>", line 488 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 995 in exec_module
  File "<frozen importlib._bootstrap>", line 935 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1331 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1360 in _find_and_load
  File "<stdin>", line 1 in <module>
Aborted (core dumped) 
```

I am not knowledgeable enough to interpret this output.

jamincollins, 2024-06-12:
I'm running into the same error, but I'm actually _trying_ to use FreeCAD within Python.  This used to work, not sure when it broke as I don't often need to leverage FreeCAD through Python.

server-bot-hanomi, 2024-06-22:
I am encountering the same problem. 

```Python 3.12.3 | packaged by Anaconda, Inc. | (main, May  6 2024, 19:46:43) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import FreeCAD
Fatal Python error: PyImport_AppendInittab: PyImport_AppendInittab() may not be called after Py_Initialize()
Python runtime state: initialized

Current thread 0x00007fa28974e440 (most recent call first):
  File "<frozen importlib._bootstrap>", line 488 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1289 in create_module
  File "<frozen importlib._bootstrap>", line 813 in module_from_spec
  File "<frozen importlib._bootstrap>", line 921 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1331 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1360 in _find_and_load
  File "<stdin>", line 1 in <module>
Aborted```

adrianinsaval, 2024-07-05:
from what you are saying it seems to me that you are not using freecad's appimage so this is not the right place to report this problem. This repo is ONLY ABOUT PACKAGING. That said this might be a problem from whatever distro you are using, I see you are using python 3.12 which AFAIK was not supported on the stable version of freecad. Please report this either to your distro or in the main FreeCAD repo with more details such as the version of freecad, the OS, installation method, etc

closed 2024-07-05 by adrianinsaval