FreeCAD/FreeCAD-Bundle #228: .AppImage is missing FreeCad Icon (no icon displayed when running)

AndKe, 2024-05-13:
Please look to the open source project PrusaSlicer to see how to add the icon to AppImage, if in doubt.

maxwxyz, 2024-05-28:
@AndKe is still present in the current weekly?

AndKe, 2024-05-28:
@maxwxyz 
 I just tested FreeCAD_weekly-builds-37543-conda-Linux-x86_64-py311.AppImage 
The issue remains.

adrianinsaval, 2024-05-28:
What icon are we talking about?

maxwxyz, 2024-05-28:
I guess the weekly AppImage icon for the application

AndKe, 2024-05-28:
I am talking about the appicon in the "task area" when the application is running.
When FreeCad AppImage is running from an AppImage, then it is represented by a default "gear" icon.
When other AppImages run, they display their own icons.

maxwxyz, 2024-05-28:
A screenshot would help but I guess you mean that how it is on my Windows:
[image]

AndKe, 2024-05-28:
[image]

A part of the vertical "taskbar/Shortcut" bar on Ubuntu  (apps with a orange dot are running)
you see a PDF reader, then PrusaSlicer , then FreeCad (the gear icon)  - it's default for iconless-apps.

adrianinsaval, 2024-05-29:
what's your system and are you running wayland or x11? If this is wayland, sorry this is a limitation of wayland and there's nothing we can do when shipping as an appimage. You can create a org.freecad.FreeCAD.destop file with the correct icon info at .local/share/applications and copy the icon to .local/share/icons to workaround this.

AndKe, 2024-05-29:
I am using x11 (mostly due to xdotools) 
I do already have a .desktop icon, (which works for the shortcut to the app(which has an icon) - but does not help once the app is started)
but unlike (for example) PrusaSlicer AppImage, FreeCAD 's appimage is missing the necessary information to make it display it's own icon when running.

AndKe, 2024-05-29:
Also: your post above me helped me because I discovered that when moving the icon to ~/.local/share/icons
, and using just a filename in the deskop file, Icon=FreeCAD.png (rather than full path to other location)   works fine.
BTW: The behavior is the same for x11 and Wayland, both works now.
The main difference is still:  Other .AppImage files dispålay their icon instantly (on both desktop managers) , but FreeCAD AppImage does not.   
Please make it work :)

adrianinsaval, 2024-05-29:
huh, that's weird I never experienced this in x11

adrianinsaval, 2024-07-05:
I cannot reproduce this problem, I can only assume it is an issue on your system. Please let me know if this can be reproduced on a clean system on x11, otherwise I'll close this issue

jose1968us, 2024-07-21:
If you check the .desktop file that is in the /tmp/.mount_freecad file generated when running freecad you will see that the reference to the icon is missing the file extension (.svg)

adrianinsaval, 2024-08-13:
> If you check the .desktop file that is in the /tmp/.mount_freecad file generated when running freecad you will see that the reference to the icon is missing the file extension (.svg)

this should not be a problem, consider reporting this bug to your DE instead

closed 2024-09-28 by adrianinsaval