FreeCAD/FreeCAD-Bundle #229: Freecad YAML parsing error all - if it is in a subfolder with a special name

Superbobo75, 2024-05-14:
If Freecad is unpacked and started from a subfolder with a special name - e.g. in Czech "_3D tiskárny" anything in Freecad stops working after starting with a YAML parsing errors:

```
YAML parsing error: 'D:/Data/_3D tiskárny/software/Freecad Dev/data/Mod/Material/Resources/Materials/Appearance/Aluminum.FCMat'
'bad file: D:/Data/_3D tiskárny/software/Freecad Dev/data/Mod/Material/Resources/Materials/Appearance/Aluminum.FCMat'
```

If I move the Freecad folder to root, it works fine.

This happens from FreeCAD_weekly-builds-37249-conda-Windows-x86_64-py311.7z

Superbobo75, 2024-05-16:
One more thing. Even if I move the Freecad folder so that the path does not contain special characters, after creating the sketch and inserting the constraints, I get an <Exception> error <Exception> TopoShapeExpansion.cpp(2032): Invalid shape name Constraint + number 

During testing I still found that (I haven't saved all released weekly versions) the versions with python 310 are fine. 
36319 - ok, ok
37213 - subfolder error, constraint ok
37249 - subfolder error, constraint ok
37302 - subfolder error, constraint error

I have the python version installed in Windows 10 Pro 22H2 19045.4291:
3.10.6 a 3.11.9

I went back to dev version 36319 for now
[image]

adrianinsaval, 2024-05-28:
This is likely not a bundle problem but a bug in FreeCAD itself

Superbobo75, 2024-05-29:
In my opinion, if it didn't do it before, and it only started from a certain package - then it's the fault of the package where the problem was introduced. Or what do you mean? In normal version 0.21.2 everything is fine too.

adrianinsaval, 2024-05-29:
the code changes in newer versions, more than the package does. Specifically the Materials system is new code. Is the latest version still showing this problem?