FreeCAD/FreeCAD-Bundle #237: Broken icon on Gnome Shell

leoheck, 2024-05-29:
This was fixed in the past, now it is broken again. 
There are 2 places where it is missing, on the bottom of the window and the favorites bar.
[image]

Here the icon appears, not sure it is the right one, but there is something.
[image]

This one is what they are using on discord
[image]

GAZ082, 2024-06-10:
In the discord has been talks of having a different DEV icon just to make sure you start the right version of the program. But I agree it should be displayed instead of the cog.

maxwxyz, 2024-06-11:
@adrianinsaval are there different locations where the separate image was used which need to point to the default icon?

leoheck, 2024-06-11:
in the past I have fixed this.. it needs the icon in some specific sizes I guess...

leoheck, 2024-06-11:
The latest version does not have this broken. However, the blue icon has gone... 

[image]

```
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * POV-Ray-Rendering
  * Assembly4 0.50.12
  * sheetmetal 0.4.17
  * InventorLoader 1.4.0
  * ose-piping
  * fasteners 0.5.21
  * asm4_disable_version_check_popup
  * OpenDark 2024.3.13
  * ProDarkThemePreferencePack 1.0.0
  * Render 2024.6.5
  * OpticsWorkbench 1.0.17
  * lattice2 1.0.0
  * kicadStepUpMod
```

maxwxyz, 2024-06-11:
is it also fixed in the app switcher? 
The icon change was intentional.

leoheck, 2024-06-13:
Yes, it is fixed there too. I am going to close this for now.

leoheck, 2024-06-13:
[image]

closed 2024-06-13 by leoheck