FreeCAD/FreeCAD-Bundle #240: Using the Subtractive Pipe can cause dependent shapes to malform

JDMils, 2024-06-03:
I'm not sure if this is to do with TNP, but it seems to be a simple shape with big problems. These are the posts I'm referring to:
https://forum.freecad.org/viewtopic.php?t=88161
https://forum.freecad.org/viewtopic.php?p=763556#p763556

Is this to do with TNP, the Subtractive Pipe or just human error or could it be a bug?

I've tested in:
`OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
`

adrianinsaval, 2024-07-05:
this is not the repo for reporting these things, this is only for packaging of the application. @bgbsww @maxwxyz is it worth transferring this to the main repo?

bgbsww, 2024-07-05:
It isn't really actionable for me in the current state.   What happens?  Does it occur in the released version?  If we are reporting a TNP problem does it happen in linkstage3?

Without that information, this would end of at the bottom of the pile until there's time to answer all those questions - and any bug reports that did provide it would tend to get attention first.

adrianinsaval, 2024-07-06:
@JDMils if this issue persists, please check if there aren't existing reports on the main repo at https://github.com/FreeCAD/FreeCAD/issues and if not create one there with details bgbsww mentioned, thanks. For now I'll close this one

closed 2024-07-06 by adrianinsaval