FreeCAD/FreeCAD-Bundle #241: Bug in face naming

FredLaCasemate, 2024-06-05:
I found an anomaly in the naming of faces when combining a fillet and a taper angle. Here are the steps to reproduce it in PartDesign:
1. Create a rectangle in a sketch and make a protrusion
2. On the upper side, create a sketch with a circle and make a protrusion. So we have:
 Pad
     Sketch
 Pad001
     Sketch001
3. On the first Pad add a taper angle (in my case, 8°).
4. Go back to the first sketch and add a fillet in a corner of the rectangle

Consequence: Sketch001 is no longer attached correctly.
[image]

GAZ082, 2024-06-10:
@FredLaCasemate please add the build version and other details by copying pasting the information from the Help menu.

Also make sure to test it with the latest dev release!

FredLaCasemate, 2024-06-10:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * AnimationFreeCAD 1.0.0
  * Assembly4 0.50.4
  * CurvedShapes 1.0.5
  * Curves 0.6.14
  * ExplodedAssembly
  * fasteners 0.4.70
  * freecad.gears 1.0.0
  * LCInterlocking
  * sheetmetal 0.3.12

maxwxyz, 2024-07-04:
Is this still an issue in the latest weekly?
If so please create an issue (if there is not already an existing one) here https://github.com/FreeCAD/FreeCAD/issues

closed 2024-07-04 by maxwxyz