FreeCAD/FreeCAD-Bundle #242: [Packaging] DMG is missing applications symlink and background

cbenhagen, 2024-06-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The current DMG for macOS is missing a symlink to the `/Applications` folder. Adding a nice background would also improve the first impression.

This is what it currently looks like:
[image]

This is how it could look like:

[image]

[image]

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * OpenDark 2024.3.13
  * Assembly4.backup1711882906.675722 0.50.1 (Disabled)
  * lattice2 1.0.0
  * fasteners.backup1711882906.7502148 0.4.56 (Disabled)
  * sheetmetal.backup1711882906.820294 0.2.63 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-06:
@adrianinsaval what do you need to implement this?

cbenhagen, 2024-06-09:
I evaluated different methods to create a nice DMG. The python package `dmgbuild` seems to be the most appropriate to use here.

1. Install with `pip install "dmgbuild[badge_icons]"`
2. Create a `settings.py` with this content:
```python
background = "builtin-arrow"
files = ["/Applications/FreeCAD.app"]
symlinks = {"Applications": "/Applications"}
badge_icon = "/Applications/FreeCAD.app/Contents/Resources/freecad.icns"
window_rect = ((100, 100), (670, 250))
icon_locations = {"FreeCAD.app": (100, 100), "Applications": (530, 100)}
```
3. `dmgbuild -s settings.py "FreeCAD" FreeCAD.dmg; open FreeCAD.dmg`

[image]

This example uses the built-in default arrow of the tool but I am sure someone will come up with a nicer background.

It looks like there are currently different scripts being used to create the [stable release](https://github.com/FreeCAD/FreeCAD/blob/main/src/Tools/macos_sign_and_notarize.sh) and the [weekly builds](https://github.com/FreeCAD/FreeCAD-Bundle/blob/main/conda/osx/create_bundle.sh). Are there plans to unify them?

cbenhagen, 2024-06-14:
Background discussion topic on Discord:
https://discord.com/channels/870877411049357352/1249312427624108032

adrianinsaval, 2024-07-05:
do we have an image for this yet? since we are already using dmgbuild now adding the image should be trivial