FreeCAD/FreeCAD-Bundle #244: Clicking new project causes Segmentation fault.

byteofsoren, 2024-06-09:
Hi I downloaded the AppImage:

`FreeCAD_weekly-builds-37668-conda-Linux-x86_64-py311.AppImage `

And when I started the program I selected dark mode and then clicked New project.

`The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
/tmp/.mount_freecaQ03xsS/AppRun: line 41: 395776 Segmentation fault      (core dumped) "${MAIN}" "$@" `

Is that a known issue?

maxwxyz, 2024-07-04:
Is this still an issue in the latest weekly?
If so please create an issue (if there is not already an existing one) here https://github.com/FreeCAD/FreeCAD/issues

closed 2024-07-04 by maxwxyz

byteofsoren, 2024-07-17:
Yes, just checked it on the latest version `FreeCAD 0.22.0, Libs: 0.22.0devR38231` (Git) and it still exists.
No, I have not created a issue on the main branch so I will do that now.
Thanks!